autonomousvision/transfuser #232: How to measure the affordances 'relative angle and lateral distance to the waypoint line

SRajasekar333, 2024-07-29:
Hi,

Kait0, 2024-08-04:
1. The data is stored [here](https://github.com/autonomousvision/transfuser/blob/fe05003765294d82f03977da40767467eca53261/team_code_autopilot/autopilot.py#L321). The angle is normalized, so you might need to unnormalize before using it.

You need to convert the ego vehicle center into the local coordinate system of the waypoint and then measure the absolute y value. We do not provide code for this you will need to write it yourself.

closed 2024-08-07 by Kait0

SRajasekar333, 2024-08-08:
Hi @Kait0,

Thanks for the reply.

1. Could you please elaborate about the normalization of the angle, how should we `unnormalize` from the value that we get from the `measurements.json`

Regards!