autonomousvision/transfuser #236: Cross-Modal Attention Statistics

MCUBE-2023, 2024-08-14:
Hi,

Based on Table 4 (Cross-Modal Attention Statistics) in your article, you are reporting the % of tokens for which at least 1 of the top-5 attended tokens belongs to the other modality for each head of the transformers: T1, T2, T3, T4.  I have this question:
Are you reporting the % of tokens for each frame, and then you are doing in average of the % for the total number of frames ? Or does this % presented in Table 4 represents only a sample from one frame ?

[image]

ap229997, 2024-08-16:
% is computed across all the frames, more details [here](https://drive.google.com/file/d/1CPI2BxvG0XC3d9DcWEl565aZqV9Mz3Q9/view).

closed 2024-08-19 by Kait0

MCUBE-2023, 2024-08-28:
@ap229997 Thank you so much!

MCUBE-2023, 2025-03-19:
Dear author, 
I have a follow-up question regarding Table 4 (Cross-Modal Attention Statistics) in your article. Are you reporting these statistics for one route (for example Route_0) ? Or are they reported for a couple of routes (for example, for Twon05 --> 32 routes) ?