autonomousvision/transfuser #238: How to install GlobalRoutePlannerDAO?

Utine, 2024-08-22:
I use carla0.9.15, Installed requirements under pythonapi folder. But I get error when run datagen.sh.
from agents.navigation.global_route_planner_dao import GlobalRoutePlannerDAO
ModuleNotFoundError: No module named 'agents.navigation.global_route_planner_dao'

Kait0, 2024-08-23:
global_route_planner_dao is part of CARLA 0.9.10.1 and was removed in newer CARLA versions.

If you want to collect data with a newer CARLA version (e.g. for CARLA leaderboard 2) I would recommend you use the [PDM-Lite expert](https://github.com/autonomousvision/carla_garage/tree/leaderboard_2) which works with carla0.9.15.

Otherwise you need to adapt the code yourself to work with CARLA 0.9.15, or downgrade if you don't need the features of newer CARLA versions.
The supported CARLA version for this repo is 0.9.10.1.

akshay18iitg, 2024-08-23:
Hi @Utine, what torch and cuda version are you using? I am unable to set the requirements.

closed 2024-08-26 by Kait0