autonomousvision/transfuser #239: Unable to recreate results by following the instructions given in the github Readme

akshay18iitg, 2024-08-24:
I am facing issue with installing torch scatter with torch 0.11.0. Can anyone assist me with the problem

suryajayaraman, 2024-08-25:
@akshay18iitg , by torch 0.11.0, do you mean torch 1.11 version or torchvision 0.12.0? Did you try creating a conda environment with the provided environment.yml file?

akshay18iitg, 2024-08-25:
I mean torch 1.11. Ya. I created conda environment with provided environmnet.yml file. I also installed torch scatter and mmcv full as mentioned in the repo. But, I am facing issues with compatibility of torch scatter.

suryajayaraman, 2024-08-26:
@akshay18iitg , did you try running the script? I remember seeing similar errors, but did not have issues running the model

akshay18iitg, 2024-08-26:
@suryajayaraman , ya I tried running local evaluation script. But it gives an error related to torch scatter

Kait0, 2024-08-26:
You can simply comment [this line](https://github.com/autonomousvision/transfuser/blob/3852e65e070f2efcfaba18fbbdb48057096b8a9e/team_code_transfuser/model.py#L11) and save yourself the hassle of compiling torch scatter (it sometimes compiles against your local cuda installation). It is only needed for a variation that you will likely not use.

akshay18iitg, 2024-08-26:
@Kait0 . Ya. It works after commenting the line. Just want to confirm one more thing. The longest6 results shared in youtube video can be replicated using the given pretrained weights?

Kait0, 2024-08-27:
yes

akshay18iitg, 2024-08-27:
Thanks @Kait0 for resolving the issue.

closed 2024-08-27 by akshay18iitg