autonomousvision/transfuser #242: Error on Sensor based model applied to datagen

StevenWICV, 2024-09-13:
Thanks for the great work.

I tried to run datagen.sh with setting below:

export CARLA_ROOT=${1:-/home/wicv/workspace/carla_0.9.10.1}
export WORK_DIR=${2:-/home/wicv/workspace/transfuser}

export CARLA_SERVER=${CARLA_ROOT}/CarlaUE4.sh
export PYTHONPATH=$PYTHONPATH:${CARLA_ROOT}/PythonAPI
export PYTHONPATH=$PYTHONPATH:${CARLA_ROOT}/PythonAPI/carla
export PYTHONPATH=$PYTHONPATH:$CARLA_ROOT/PythonAPI/carla/dist/carla-0.9.10-py3.7-linux-x86_64.egg
export SCENARIO_RUNNER_ROOT=${WORK_DIR}/scenario_runner
export LEADERBOARD_ROOT=${WORK_DIR}/leaderboard
export PYTHONPATH="${CARLA_ROOT}/PythonAPI/carla/":"${SCENARIO_RUNNER_ROOT}":"${LEADERBOARD_ROOT}":${PYTHONPATH}

export SCENARIOS=${WORK_DIR}/leaderboard/data/training/scenarios/Scenario10/Town10HD_Scenario10.json
export ROUTES=${WORK_DIR}/leaderboard/data/training/routes/Scenario10/Town10HD_Scenario10.xml
export REPETITIONS=1
export CHALLENGE_TRACK_CODENAME=SENSORS
export CHECKPOINT_ENDPOINT=${WORK_DIR}/results/tfuse_Town10HD_Scenario10.json
export SAVE_PATH=${WORK_DIR}/recordRunning/tfuse_eval/Town10HD_Scenario10
export TEAM_AGENT=${WORK_DIR}/team_code_transfuser/submission_agent.py
export TEAM_CONFIG=${WORK_DIR}/model_ckpt/transfuser
export DEBUG_CHALLENGE=0
export RESUME=0
export DATAGEN=1

python3 ${LEADERBOARD_ROOT}/leaderboard/leaderboard_evaluator_local.py \
--scenarios=${SCENARIOS}  \
--routes=${ROUTES} \
--repetitions=${REPETITIONS} \
--track=${CHALLENGE_TRACK_CODENAME} \
--checkpoint=${CHECKPOINT_ENDPOINT} \
--agent=${TEAM_AGENT} \
--agent-config=${TEAM_CONFIG} \
--debug=${DEBUG_CHALLENGE} \
--resume=${RESUME}

---------------------------------------------------------------------------------

I got error:

Could not set up the required agent:
> __init__() takes 2 positional arguments but 3 were given

Traceback (most recent call last):
  File "/home/wicv/workspace/transfuser/leaderboard/leaderboard/leaderboard_evaluator_local.py", line 270, in _load_and_run_scenario
    self.agent_instance = getattr(self.module_agent, agent_class_name)(args.agent_config, config.index)
TypeError: __init__() takes 2 positional arguments but 3 were given

---------------------------------------------------------------------------------

How can I figure this out? 

Many thanks

Kait0, 2024-09-13:
You want to collect data with transfuser?
Usually you use https://github.com/autonomousvision/transfuser/blob/2022/team_code_autopilot/data_agent.py

If you want to collect data with transfuser you need to change the  [inheritance](https://github.com/autonomousvision/transfuser/blob/3852e65e070f2efcfaba18fbbdb48057096b8a9e/team_code_transfuser/submission_agent.py#L34C19-L34C35) to [autonomous_agent_local](https://github.com/autonomousvision/transfuser/blob/3852e65e070f2efcfaba18fbbdb48057096b8a9e/team_code_autopilot/autopilot.py#L24).

StevenWICV, 2024-09-14:
> You want to collect data with transfuser? Usually you use https://github.com/autonomousvision/transfuser/blob/2022/team_code_autopilot/data_agent.py
>
> If you want to collect data with transfuser you [...]

Really appreciate your quick reply.

Basically, I'd like to generate images from sensors like RGB or topdown to get more details of running scenario. Cause Carla only show BEV when running script.

I have tried two inheritance settings 

----------------------------------------------------
class HybridAgent(autonomous_agent_local.AutonomousAgent):
    def setup(self, path_to_conf_file, route_index=None):
        self.track = autonomous_agent_local.Track.SENSORS

Got Error:

The sensor's configuration used is invalid:
> You are submitting to the wrong track [Track.SENSORS]!

Traceback (most recent call last):
  File "/home/wicv/workspace/transfuser/leaderboard/leaderboard/leaderboard_evaluator_local.py", line 280, in _load_and_run_scenario
    AgentWrapper.validate_sensor_configuration(self.sensors, track, args.track)
  File "/home/wicv/workspace/transfuser/leaderboard/leaderboard/autoagents/agent_wrapper_local.py", line 203, in validate_sensor_configuration
    raise SensorConfigurationInvalid("You are submitting to the wrong track [{}]!".format(Track(selected_track)))
leaderboard.envs.sensor_interface.SensorConfigurationInvalid: You are submitting to the wrong track [Track.SENSORS]!

----------------------------------------------------
class HybridAgent(autonomous_agent_local.AutonomousAgent):
    def setup(self, path_to_conf_file, route_index=None):
        self.track = autonomous_agent.Track.SENSORS

Got Error:

The scenario could not be loaded:
> 'rotation_frequency'

Traceback (most recent call last):
  File "/home/wicv/workspace/transfuser/leaderboard/leaderboard/leaderboard_evaluator_local.py", line 329, in _load_and_run_scenario
    self.manager.load_scenario(scenario, self.agent_instance, config.repetition_index)
  File "/home/wicv/workspace/transfuser/leaderboard/leaderboard/scenarios/scenario_manager_local.py", line 116, in load_scenario
    self._agent.setup_sensors(self.ego_vehicles[0], self._debug_mode)
  File "/home/wicv/workspace/transfuser/leaderboard/leaderboard/autoagents/agent_wrapper_local.py", line 128, in setup_sensors
    bp.set_attribute('rotation_frequency', str(sensor_spec['rotation_frequency']))
KeyError: 'rotation_frequency'
----------------------------------------------------

Kait0, 2024-09-14:
In datagen mode you need to manually specify [these LiDAR paramters](https://github.com/autonomousvision/transfuser/blob/3852e65e070f2efcfaba18fbbdb48057096b8a9e/leaderboard/leaderboard/autoagents/agent_wrapper_local.py#L131). You can set them to the same as the default.

If you just want to change the spectator camera of CARLA it might be easier to just adjust the parameters [here](https://github.com/autonomousvision/transfuser/blob/3852e65e070f2efcfaba18fbbdb48057096b8a9e/leaderboard/leaderboard/scenarios/scenario_manager.py#L177). Or use the visualization function of transfuser by setting the environment variable [SAVE_PATH](https://github.com/autonomousvision/transfuser/blob/3852e65e070f2efcfaba18fbbdb48057096b8a9e/team_code_transfuser/submission_agent.py#L23) and [debug=True](https://github.com/autonomousvision/transfuser/blob/3852e65e070f2efcfaba18fbbdb48057096b8a9e/team_code_transfuser/config.py#L35C5-L35C11).