autonomousvision/transfuser #244: Evaluate the model on WSL

Yanchen30247, 2024-09-23:
Thank you for your excellent work on automomous driving. I am trying to Evaluate the model on WSL2, however, I tried many times but the models did not work well.

I am running Carla server on my windows system and try to operate the client on ubuntu(WSL). The server connection is good but seems the agent couldn't set up.
[image]

The result also could be generated but from the evaluate result, it shows that Agent couldn't be set up
[transfuser_longest6.json](https://github.com/user-attachments/files/17099365/transfuser_longest6.json)

closed 2024-09-23 by Yanchen30247

Kait0, 2024-09-23:
Is solved?
These warnings about version mismatch should not appear.
Perhaps you are using the wrong CARLA version. (should be 0.9.10.1).
Also you need to make sure your windows subsystem for linux actually supports / cuda / GPU access. Used to be tricky in the past not sure what the current status of the technology is.
If there is a problem with setting up the agent then you should usually get a more detailed stacktrace.

Kait0, 2024-09-23:
If the agent could not be setup error occurs then the code usually prints a stacktrace, see [here](https://github.com/autonomousvision/transfuser/blob/3852e65e070f2efcfaba18fbbdb48057096b8a9e/leaderboard/leaderboard/leaderboard_evaluator_local.py#L305C30-L305C54). Bit weird that I don't see one in your screenshot.

We typically add the carla client .egg file via an [environment variable](https://github.com/autonomousvision/transfuser/blob/3852e65e070f2efcfaba18fbbdb48057096b8a9e/leaderboard/scripts/local_evaluation.sh#L7). Perhaps there is another one in your env from a different project that gets loaded instead.
It is also possible to install the .egg file into pip/conda env but you need to first downgrade setuptools to an older version (maybe 2020 or so) because .egg files are deprecated nowadays (can upgrade setuptools again after installing).

Yanchen30247, 2024-09-23:
> You might want to delete the transfuser_longest6.json file and run again to get a better error message. Could be that the crash was in an older run and now the software just terminates because all routes are already in the transfuser_longest6.json file.

Hi, Thank you for your help! I solved the problem by changing the transfuser\leaderboard\scripts\local_evaluation.sh 

I delete the line of "export CARLA_SERVER=${CARLA_ROOT}/CarlaUE4.sh" and the code runs 

However, the angle of view is in my carla is a little bit strange
[image]
Can I change the view into a third-person view or fisrt-person view?