autonomousvision/transfuser #246: Low FPS and camera twitch

Yanchen30247, 2024-10-23:
Hi, I appreciate your great work! But when I tried to run and evaluate the pre-trained model on my PC. I faced low FPS and camera twitch, the running FPS is lower than 10 FPS while facing complex situation. 

[image]

Also the camera faced frequently twitch during the running. (Means when the vehicle is accelerating, the camera would have a sharp backward blow)
[image]

Kait0, 2024-10-23:
Simulation around 10 FPS is not that unusual I think.
You can measure but TF usually uses in the order of 50ms, and you needs to add CARLA simulation on top of that, which can easily go in the range of 100ms / 10 FPS.
TransFuser is a comparatively fast architecture since it is single frame and uses low resolution cameras.

You second problem is also normal if what you show is the spectator camera.
The spectator camera is not attached to the vehicle and is readjusted after the vehicle moved leading to this twitch.
This should not happen with the cameras mounted on the car.

Yanchen30247, 2024-10-25:
> Simulation around 10 FPS is not that unusual I think. You can measure but TF usually uses in the order of 50ms, and you needs to add CARLA simulation on top of that, which can easily go in the range [...]

Thank you for your reply, I appreciate your reply. I am just wondering if I can use more powerful CPU to get higher FPS. AND for my second question, I think the camera in your demo.gif is very stable, did you use the attached camera on car?

Kait0, 2024-10-25:
CPU performance has not increased much in the recent years and you already seem to have a very good CPU so probably not. Maybe you can buy a better CPU but you also already have a very good GPU so, you would need a server grade GPU.
Don't think that is worth it.

You can use the build in [visualization function](https://github.com/autonomousvision/transfuser/blob/3852e65e070f2efcfaba18fbbdb48057096b8a9e/team_code_transfuser/model.py#L726), to generate stable frames and then convert them to a video with ffmpeg or so. 

You need to set debug and save_path to use the visualization [here](https://github.com/autonomousvision/transfuser/blob/3852e65e070f2efcfaba18fbbdb48057096b8a9e/team_code_transfuser/model.py#L726).

closed 2024-10-25 by Kait0