autonomousvision/transfuser #247: extract the model from the Leaderboard env and run it in a simple asynchronous env

StevenWICV, 2024-10-28:
Thanks for your great work.

I'd like to extract the model from the Leaderboard env and run it in a simple asynchronous Carla env in order to load multiple transfuser control drivers on the map. 

Could you please give some suggestions or example to do this?

Kait0, 2024-10-28:
Hm I never used CARLA in asynchronous mode. I suspect you would need to modify the carla leaderboard code, or write your own asynchronous client.
The main code of transfuser is [here](https://github.com/autonomousvision/transfuser/blob/2022/team_code_transfuser/submission_agent.py).
Some of the code like the A* planner are part of the leaderboard, you would need to extract it or use an adapted leaderboard.

StevenWICV, 2024-10-29:
Appreciate your reply. I'll try it.

closed 2024-10-29 by Kait0