autonomousvision/transfuser #248: Number of routes in Town05

MCUBE-2023, 2024-11-06:
Hi,

Would you please tell me how many routes are they in total in Town05 ? And are they long or short routes ?

In both cases (long routes of Town05 or short routes of Town05), would you please refer me to the directory and (xml or json files) regarding town05 and its routes ?

Kait0, 2024-11-06:
Town 05 short is basically solved by now I think, the reported state-of-the-art  is 90+.
Town 05 long is more challenging because the driving score increases in difficulty with route length, so if you are trying to push state-of-the-art it is more suitable.

Note that the town05 long benchmark uses [scenarios](https://github.com/autonomousvision/transfuser/blob/cvpr2021/leaderboard/data/scenarios/town05_all_scenarios.json) (people sometime "forget" to use them when reporting this benchmark). And the routes are meant as validation routes, so you are not allowed to train on data collected in Town 05 (also saw a paper that forgot this).
Lastly you should run at least 3 repetitions of the routes to reduce the impact of the random seed on your result.

MCUBE-2023, 2024-11-06:
Thanks for your quick reply, 

Sorry, but I am still confused. To benchmark consistently an experience, noting that I am using a single GPU. Should I use the validation routes or evaluation routes of Town05 ?

Kait0, 2024-11-07:
The folder names for validation routes or evaluation routes have no meaning. The names of the files is what matters.
Both town_05_long and town_05_short are official benchmarks see Table 1 of the [paper](https://arxiv.org/pdf/2104.09224).

If you only have a single GPU Town 05 short might be better because it requires less compute / time to evaluate.

MCUBE-2023, 2024-11-07:
That's crystal clear! Thank you so much for answering accurately my question :)

closed 2024-11-07 by MCUBE-2023