autonomousvision/transfuser #250: Question about Submitting to the CARLA leaderboard

MCUBE-2023, 2024-11-09:
Dear Author,
I have 3 questions related to submitting to the CARLA leaderboard. Following your GitHub instruction, I made sure to adapt adequately the paths in [make_docker.sh](https://github.com/autonomousvision/transfuser/blob/2022/leaderboard/scripts/make_docker.sh). 

Then I run these lines:

cd leaderboard
cd scripts
./make_docker.sh

The execution stops at Step 11/67 , and it asks me to select my Geographic area. So I typed the number 2 which corresponds to the America area, As you can see in **the attached screen-shot,** I got the cursor which I pointed with a red arrow, and this cursor is not even blinking. I hit the key enter, but the cursor is just moving to the next line without even blinking. I don't know if there's something running in the back-end, but I was waiting for 30 minutes moment and nothing happened. 

And this is the output that I got on my terminal (I am copying and pasting what I got on my terminal):

-------------------------------------------------------------------------------------------------------------
mv: target '.tmp/PythonAPI/carla/dist/carla-leaderboard-py2.7.egg' is not a directory
mv: target '.tmp/PythonAPI/carla/dist/carla-leaderboard-py3x.egg' is not a directory
Sending build context to Docker daemon  4.886GB
Step 1/67 : FROM nvidia/cuda:12.1.1-cudnn8-devel-ubuntu20.04
 ---> 15ca36fdea07
Step 2/67 : ARG HTTP_PROXY
 ---> Using cache
 ---> 8c51fff569cb
Step 3/67 : ARG HTTPS_PROXY
 ---> Using cache
 ---> a313cdcd0bc3
Step 4/67 : ARG http_proxy
 ---> Using cache
 ---> 1e8fb229ff76
Step 5/67 : RUN apt-get update && apt-get install --reinstall -y locales && locale-gen en_US.UTF-8
 ---> Using cache
 ---> 90ad0eecb4ed
Step 6/67 : ENV LANG en_US.UTF-8
 ---> Using cache
 ---> c06866241918
Step 7/67 : ENV LANGUAGE en_US
 ---> Using cache
 ---> 7ac0d601f0f2
Step 8/67 : ENV LC_ALL en_US.UTF-8
 ---> Using cache
 ---> 3e844b2b6839
Step 9/67 : RUN apt-get update && apt-get install -y --no-install-recommends
 ---> Using cache
 ---> baeee0cffbd6
Step 10/67 : RUN apt-get install -y build-essential
 ---> Using cache
 ---> 4e17afae4ce3
Step 11/67 : RUN apt-get install -y cmake
 ---> Running in 90949e7797bd
Reading package lists...
Building dependency tree...
Reading state information...
The following additional packages will be installed:
  cmake-data krb5-locales libarchive13 libbrotli1 libcurl4 libexpat1
  libgssapi-krb5-2 libicu66 libjsoncpp1 libk5crypto3 libkeyutils1 libkrb5-3
  libkrb5support0 libnghttp2-14 libpsl5 librhash0 librtmp1 libssh-4 libuv1
  libxml2 publicsuffix tzdata
Suggested packages:
  cmake-doc ninja-build lrzip krb5-doc krb5-user
The following NEW packages will be installed:
  cmake cmake-data krb5-locales libarchive13 libbrotli1 libcurl4 libexpat1
  libgssapi-krb5-2 libicu66 libjsoncpp1 libk5crypto3 libkeyutils1 libkrb5-3
  libkrb5support0 libnghttp2-14 libpsl5 librhash0 librtmp1 libssh-4 libuv1
  libxml2 publicsuffix tzdata
0 upgraded, 23 newly installed, 0 to remove and 56 not upgraded.
Need to get 17.0 MB of archives.
After this operation, 74.5 MB of additional disk space will be used.
Get:1 http://archive.ubuntu.com/ubuntu focal-updates/main amd64 libexpat1 amd64 2.2.9-1ubuntu0.7 [74.7 kB]
Get:2 http://archive.ubuntu.com/ubuntu focal-updates/main amd64 tzdata all 2024a-0ubuntu0.20.04.1 [301 kB]
Get:3 http://archive.ubuntu.com/ubuntu focal-updates/main amd64 libicu66 amd64 66.1-2ubuntu2.1 [8,515 kB]
Get:4 http://archive.ubuntu.com/ubuntu focal-updates/main amd64 libxml2 amd64 2.9.10+dfsg-5ubuntu0.20.04.7 [640 kB]
[... 96 lines cut ...]
Setting up krb5-locales (1.17-6ubuntu4.7) ...
Setting up libkrb5support0:amd64 (1.17-6ubuntu4.7) ...
Setting up tzdata (2024a-0ubuntu0.20.04.1) ...
debconf: unable to initialize frontend: Dialog
debconf: (TERM is not set, so the dialog frontend is not usable.)
debconf: falling back to frontend: Readline
Configuring tzdata
------------------

Please select the geographic area in which you live. Subsequent configuration
questions will narrow this down by presenting a list of cities, representing
the time zones in which they are located.

  1. Africa      4. Australia  7. Atlantic  10. Pacific  13. Etc
  2. America     5. Arctic     8. Europe    11. SystemV
  3. Antarctica  6. Asia       9. Indian    12. US
Geographic area: 
-----------------------------------------------------------------------------------------------------------

**So these are the 3 questions I am asking you please: 
1- Is there something abnormal in the execution based on the part that you are seeing and which I copied and pasted from my terminal ?
2- After selecting my geographic area, and assuming that something is running at the backend (the cursor is not even blinking), how long should I wait after selecting my geographic area before proceeding to the next step?
3- In case of running this script from Canada, should I select the option America ? And in this case, should type the number 2 or should I write this option in plain text (like typing the word America)**

[image]

Kait0, 2024-11-11:
You should not have to enter anything when making a docker container.
Perhaps you need to add some environment variables to turn the interactive behavior off.
Maybe you can have a look [here](https://askubuntu.com/questions/909277/avoiding-user-interaction-with-tzdata-when-installing-certbot-in-a-docker-contai).

Note that the CARLA leaderboard has changed servers so the docu in our repo is outdated.
You can find up to date version on their [website](https://leaderboard.carla.org/get_started_v1/).
As far as I know the leaderboard servers on eval.ai are closed currently though.

MCUBE-2023, 2024-11-13:
Thank you for your help. I adapted the Dockerfile.master to turn the interactive behaviour off. And this is the output that I have on my terminal after running./make_docker.sh:

Sending build context to Docker daemon  2.817GB
Step 1/40 : FROM nvidia/cuda:12.1.1-cudnn8-devel-ubuntu20.04
 ---> 15ca36fdea07
Step 2/40 : ENV DEBIAN_FRONTEND=noninteractive
 ---> Using cache
 ---> 04554d9439bf
Step 3/40 : ARG HTTP_PROXY
 ---> Using cache
 ---> f8de7a714de0
Step 4/40 : ARG HTTPS_PROXY
 ---> Using cache
 ---> a3d28de064d2
Step 5/40 : ARG http_proxy
 ---> Using cache
 ---> cf86644e96c3
Step 6/40 : RUN apt-get update && apt-get install --reinstall -y locales && locale-gen en_US.UTF-8
 ---> Using cache
 ---> 098e4b959120
Step 7/40 : ENV LANG en_US.UTF-8
 ---> Using cache
 ---> e700db69dcc6
Step 8/40 : ENV LANGUAGE en_US
 ---> Using cache
 ---> 5bee64139459
Step 9/40 : ENV LC_ALL en_US.UTF-8
 ---> Using cache
 ---> 3504e6635e4d
Step 10/40 : RUN apt-get update &&     apt-get install -y --no-install-recommends         build-essential cmake git curl vim         ca-certificates libjpeg-dev libpng16-16 libtiff5 libpng-dev         python3-dev python3.8 python3.8-dev python3-pip python3-setuptools     && pip3 install --upgrade pip==20.3.4 &&     rm -rf /var/lib/apt/lists/*
 ---> Running in 03b56809bc64
Hit:1 http://archive.ubuntu.com/ubuntu focal InRelease
Get:2 http://archive.ubuntu.com/ubuntu focal-updates InRelease [128 kB]
Hit:3 http://security.ubuntu.com/ubuntu focal-security InRelease
Get:4 http://archive.ubuntu.com/ubuntu focal-backports InRelease [128 kB]
Hit:5 https://developer.download.nvidia.com/compute/cuda/repos/ubuntu2004/x86_64  InRelease
Fetched 256 kB in 1s (509 kB/s)
Reading package lists...
Reading package lists...
Building dependency tree...
Reading state information...
build-essential is already the newest version (12.8ubuntu1.1).
build-essential set to manually installed.
The following additional packages will be installed:
  cmake-data git-man libarchive13 libasound2 libasound2-data libbrotli1
  libcanberra0 libcurl3-gnutls libcurl4 liberror-perl libexpat1 libexpat1-dev
  libgpm2 libgssapi-krb5-2 libicu66 libjbig0 libjpeg-turbo8 libjpeg-turbo8-dev
  libjpeg8 libjpeg8-dev libjsoncpp1 libk5crypto3 libkeyutils1 libkrb5-3
  libkrb5support0 libltdl7 libmpdec2 libnghttp2-14 libogg0 libpsl5
  libpython3-dev libpython3-stdlib libpython3.8 libpython3.8-dev
  libpython3.8-minimal libpython3.8-stdlib librhash0 librtmp1 libssh-4 libtdb1
  libuv1 libvorbis0a libvorbisfile3 libwebp6 libxml2 mime-support
  python-pip-whl python3 python3-distutils python3-lib2to3 python3-minimal
  python3-pkg-resources python3-wheel python3.8-minimal
  sound-theme-freedesktop tzdata vim-common vim-runtime xxd zlib1g-dev
Suggested packages:
  cmake-doc ninja-build gettext-base git-daemon-run | git-daemon-sysvinit
  git-doc git-el git-email git-gui gitk gitweb git-cvs git-mediawiki git-svn
  lrzip libasound2-plugins alsa-utils libcanberra-gtk0 libcanberra-pulse gpm
  krb5-doc krb5-user python3-doc python3-tk python3-venv python-setuptools-doc
  python3.8-venv python3.8-doc binfmt-support ctags vim-doc vim-scripts
Recommended packages:
  less ssh-client alsa-ucm-conf alsa-topology-conf krb5-locales libpng-tools
  publicsuffix file
The following NEW packages will be installed:
  cmake cmake-data curl git git-man libarchive13 libasound2 libasound2-data
  libbrotli1 libcanberra0 libcurl3-gnutls libcurl4 liberror-perl libexpat1
  libexpat1-dev libgpm2 libgssapi-krb5-2 libicu66 libjbig0 libjpeg-dev
  libjpeg-turbo8 libjpeg-turbo8-dev libjpeg8 libjpeg8-dev libjsoncpp1
  libk5crypto3 libkeyutils1 libkrb5-3 libkrb5support0 libltdl7 libmpdec2
[... 2292 lines cut ...]
 ---> 17bb9f527418
Step 37/40 : ENV HTTPS_PROXY ""
 ---> Running in 498b8c7f4ab9
Removing intermediate container 498b8c7f4ab9
 ---> 97f04e3fda33
Step 38/40 : ENV http_proxy ""
 ---> Running in 4ab69f732baa
Removing intermediate container 4ab69f732baa
 ---> 9ebc65c64225
Step 39/40 : ENV https_proxy ""
 ---> Running in 42823231f30c
Removing intermediate container 42823231f30c
 ---> 1f8ad1565b60
Step 40/40 : CMD ["/bin/bash"]
 ---> Running in 845525cb3e39
Removing intermediate container 845525cb3e39
 ---> a11615a206ff
Successfully built a11615a206ff
Successfully tagged transfuser-agent:latest

**So I have these 2 questions:

1- Based on what I have copied and pasted from my terminal, can I presume that make_docker.sh has been executed correctly since he completed succussfully the last step which is 40/40 ?

2- What is the generated output after the successful and complete execution of this script? I am quite confused about the term 'docker image', so if I am understanding things correctly, I am supposed to have a docker image generated after running that script? Right?  And I presume that this docker image is a portable application? Right? And what is the root of the generated docker image ?**

Kait0, 2024-11-14:
Yes looks ok.
Docker images aren't really individual files. I don't know how docker. works internally but I think there are distributed somewhere in the docker folders.
It is possible to export a docker image into an individual file, but you usually don't need to do this.

You can test your docker container using [this script](https://github.com/autonomousvision/transfuser/blob/2022/tools/run_docker.sh).
It will log you into a bash terminal inside the docker container. 
From there you can try to start the leaderboard_evaluator with python. The CARLA server can be started from outside the docker container (it can communicate with software run inside the container).
If transfuser runs this way you can be pretty certain that everything went right.

MCUBE-2023, 2024-11-15:
Thank you for your response.

I followed the steps to submit a docker image on Carla Leaderboard following the steps in this link: https://leaderboard.carla.org/submit/

And as illustrated in the first screenshot, I reached the step 5 which is 'submit'. However, as you can see in the 2nd screenshot, it is telling me and as highlighted in red that "In order to participate in the Leaderboard 2.0 - SENSORS Track and Leaderboard 2.0 - MAP Track, a minimum score will be needed in their respective qualifier tracks, which consist of a short route with simple scenarios". 

So, honestly, I am quite confused about the next steps. What should I do for the next step?

[image]
[image]

Kait0, 2024-11-15:
To submit to the leaderboard your team needs to be approved by the CARLA challenge organizers.
It is important to add your institutional affiliation for that. You can message them on e-mail or discord if they take too long to approve.

Next, the leaderboard has a short qualifying stage to prevent many bad solutions wasting compute on the expensive evaluation.
You need to first submit to the qualifier and pass a minimum score in order to be eligible to submit to the leaderboard.
Instructions on how to do that are also on eval.ai.

closed 2024-11-21 by Kait0