autonomousvision/transfuser #251: Regarding Training Dataset

akshay18iitg, 2024-11-14:
I am planning to collect my own dataset using your data collection scripts. How much time is required to collect a dataset of similar size to the original dataset when running autopilot on a single A100 GPU?

Kait0, 2024-11-15:
Hm maybe something in the range of 300-400 GPU hours.
When we collect the dataset we usually use around 32x 2080ti GPUs and that can finish within a day.
If you have access to only a single big GPU, you can probably speed up the collection by running multiple CARLA servers on one GPU (you will need to write some scripts for that).
This could give you linear speedups but you need a good enough CPU with many cores (most processing is CPU I think).
So maybe its possible to collect it in 2 days if you run 8-16 CARLA servers on the GPU, but I have never tried that.

akshay18iitg, 2024-11-15:
I checked my GPU and CPU specifications. 
I have one 32 CPU cores and one 16GB VRAM Quadro GPU. 
Will this be sufficient to run 8 Carla instance?

akshay18iitg, 2024-11-15:
Also, can you provide the code you have written for collecting data on multiple GPU

Kait0, 2024-11-16:
Probably not CARLA requires between 4 and 8 GB of VRAM per instance depending on which version you use.

We released a script to collect a dataset on a SLURM cluster in the TransFuser++ repository, you can have a look [here](https://github.com/autonomousvision/carla_garage/blob/main/generate_dataset_slurm.py).

akshay18iitg, 2024-11-16:
Thanks for helping. 
I just checked that using -RenderOffScreen command helps Carla to use only 1 - 2 GB VRAM per instance.

closed 2024-11-16 by akshay18iitg

zzhzz4429, 2024-11-18:
Hello,

Thank you for your fantastic work on this project. I have a question regarding the data collection time mentioned in your paper. You mentioned that the autopilot stored data at 2 FPS (frames per second), and the original dataset contains 228,000 data points. Based on this, I calculated that collecting 228,000 data points at 2 FPS would take approximately 31.6 hours. However, you stated that data collection took 300-400 hours. Could you please explain why there is such a significant difference in practice?

Kait0, 2024-11-18:
The 2 FPS refer to simulation time which is different from wall clock time.
In CARLA synchronous mode, every tick always represents 50 ms simulation time (depending on simulator settings) regardless of how long it took to compute.
See the [docs](https://carla.readthedocs.io/en/0.9.10/adv_synchrony_timestep/) for a more in depth explanation.

You can estimate how long it would take to collect your dataset by running data collection on a route an measuring how long it took per frame. We are saving every 5th frame, so you need to run roughly 1.140.000 CARLA steps.

zzhzz4429, 2024-11-19:
> The 2 FPS refer to simulation time which is different from wall clock time. In CARLA synchronous mode, every tick always represents 50 ms simulation time (depending on simulator settings) regardless [...]

Thanks for the timely respond!

Kait0, 2024-11-19:
actually I was thinking about TF++ which saves at 4 Hz. TransFuser was at  2Hz so every 10th frame -> 2.280.000 CARLA steps.