autonomousvision/transfuser #252: Version Mismatch for CARLA

st3lzer, 2024-11-19:
Hello,

I have a question related to this repository and the CARLA simulator. When starting data generation and evaluation, I receive the following warning:

```
WARNING: Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API
WARNING: Client API version     = 784d9b9f
WARNING: Simulator API version  = 0.9.10.1
```

Despite this warning, everything runs smoothly, and the benchmark results are as expected. I’m curious about the cause of this warning.

I followed the installation guide and used the scripts from this repository, and my Docker containers are running CARLA version 0.9.10.1. I have two possible explanations for the mismatch warning:

- The client API has a slightly different Git hash, which could trigger the mismatch.
- The egg file is labeled version 0.9.10 (instead of 0.9.10.1), as there is no specific egg file for version 0.9.10.1.

Has anyone encountered this issue or have insights into the source of the warning?

Thank you!

Kait0, 2024-11-19:
You can debug [here](https://github.com/autonomousvision/transfuser/blob/3852e65e070f2efcfaba18fbbdb48057096b8a9e/leaderboard/leaderboard/leaderboard_evaluator_local.py#L90) to see why this happens. Don't think it's too important though if you have the correct simulator version.

st3lzer, 2024-11-19:
The 784d9b9f is a commit shortly after the release of CARLA 0.9.10.1 (https://github.com/carla-simulator/carla/compare/0.9.10.1...master).
This means the version downloaded by the script setup_carla.sh contains the master branch from the 2nd of October 2024, containing a few after release changes.

Is there a reason for using this version or was it just the current master branch status at the time?

Kait0, 2024-11-19:
Hm the script is just supposed to download the official CARLA release version.
The CARLA team changed the host server at some point from AWS to backblazeb2, i thought they hosted the same version, but perhaps not?
There seems to be a new host link on [GitHub](https://github.com/carla-simulator/carla/releases/tag/0.9.10.1) now. Perhaps you can try that download and see if the version hash matches exactly.

closed 2024-11-24 by Kait0