autonomousvision/transfuser #253: TypeError: zip argument #2 must support iteration

jacksonmygo, 2024-11-23:
Here is the error I met when I tried to run the result_parser.py:
(car) PS D:\Anaconda\envs\car\WindowsNoEditor\transfuser-2022\tools> python result_parser.py --xml D:\Anaconda\envs\car\WindowsNoEditor\transfuser-2022\leaderboard\data\longest6\longest6.xml --results D:\Anaconda\envs\car\WindowsNoEditor\transfuser-2022\results\autopilot_longest6.json --save_dir D:\Anaconda\envs\car\WindowsNoEditor\transfuser-2022\output --town_maps D:\Anaconda\envs\car\WindowsNoEditor\transfuser-2022\leaderboard\data\town_maps_xodr
result_parser.py:198: RuntimeWarning: invalid value encountered in double_scalars
  total_score_values = total_score_values.sum(axis=0)/len(route_evaluation)
Traceback (most recent call last):
  File "result_parser.py", line 373, in <module>
    main()
  File "result_parser.py", line 226, in main
    total_score_info = [{"label":label, "value":value} for label,value in zip(total_score_labels,total_score_values)]
TypeError: zip argument #2 must support iteration

Could you please tell me how to correct this error?

Kait0, 2024-11-24:
The --results argument should point to a folder containing .json files (it will search for them), no the file itself.
Also you should check the content of autopilot_longest6.json, there might be routes in there where a software crash has occured.

jacksonmygo, 2024-11-24:
OK, I have changed it. It works well, thank you a lot!

closed 2024-11-24 by jacksonmygo