autonomousvision/transfuser #254: AttributeError: 'LeaderboardEvaluator' object has no attribute 'manager'

jacksonmygo, 2024-11-24:
When I tried to run the local_evaluation.sh, I met the problem shown below:
(car) PS D:\Anaconda\envs\car\WindowsNoEditor\transfuser-2022\leaderboard\scripts> sh local_evaluation.sh D:\Anaconda\envs\car\WindowsNoEditor D:\Anaconda\envs\car\WindowsNoEditor\transfuser-2022
Traceback (most recent call last):
  File "D:\Anaconda\envs\car\WindowsNoEditor\transfuser-2022/leaderboard/leaderboard/leaderboard_evaluator_local.py", line 466, in main
    leaderboard_evaluator = LeaderboardEvaluator(arguments, statistics_manager)
  File "D:\Anaconda\envs\car\WindowsNoEditor\transfuser-2022/leaderboard/leaderboard/leaderboard_evaluator_local.py", line 87, in __init__
    dist = pkg_resources.get_distribution("carla")
  File "D:\Anaconda\envs\car\lib\site-packages\pkg_resources\__init__.py", line 471, in get_distribution
    dist = get_provider(dist)
  File "D:\Anaconda\envs\car\lib\site-packages\pkg_resources\__init__.py", line 347, in get_provider
    return working_set.find(moduleOrReq) or require(str(moduleOrReq))[0]
  File "D:\Anaconda\envs\car\lib\site-packages\pkg_resources\__init__.py", line 891, in require
    needed = self.resolve(parse_requirements(requirements))
  File "D:\Anaconda\envs\car\lib\site-packages\pkg_resources\__init__.py", line 777, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'carla' distribution was not found and is required by the application
Exception ignored in: <function LeaderboardEvaluator.__del__ at 0x000001C4B3523048>
Traceback (most recent call last):
  File "D:\Anaconda\envs\car\WindowsNoEditor\transfuser-2022/leaderboard/leaderboard/leaderboard_evaluator_local.py", line 122, in __del__
    self._cleanup()
  File "D:\Anaconda\envs\car\WindowsNoEditor\transfuser-2022/leaderboard/leaderboard/leaderboard_evaluator_local.py", line 134, in _cleanup
    if self.manager and self.manager.get_running_status() \
AttributeError: 'LeaderboardEvaluator' object has no attribute 'manager'
Traceback (most recent call last):
  File "D:\Anaconda\envs\car\WindowsNoEditor\transfuser-2022/leaderboard/leaderboard/leaderboard_evaluator_local.py", line 476, in <module>
    main()
  File "D:\Anaconda\envs\car\WindowsNoEditor\transfuser-2022/leaderboard/leaderboard/leaderboard_evaluator_local.py", line 472, in main
    del leaderboard_evaluator
UnboundLocalError: local variable 'leaderboard_evaluator' referenced before assignment

Could you please tell me how can I solve these three problems shown above? Thanks a lot!

Kait0, 2024-11-26:
This is only one problem:

> File "D:\Anaconda\envs\car\WindowsNoEditor\transfuser-2022/leaderboard/leaderboard/leaderboard_evaluator_local.py", line 87, in init
> dist = pkg_resources.get_distribution("carla")
>  pkg_resources.DistributionNotFound: The 'carla' distribution was not found and is required by the application

Typically CARLA is loaded by appending its .egg file to the [python path](https://github.com/autonomousvision/transfuser/blob/3852e65e070f2efcfaba18fbbdb48057096b8a9e/leaderboard/scripts/local_evaluation.sh#L7).

jacksonmygo, 2024-11-26:
Thank you for your reply! I changed the Python path according to my file, but it still met the same problem "pkg_resources.DistributionNotFound: The 'carla' distribution was not found and is required by the application". The whole content of my local_evaluation.sh is shown below:
export CARLA_ROOT=${1:-/home/kchitta/Documents/CARLA_0.9.10.1}
export WORK_DIR=${2:-/home/kchitta/Documents/transfuser}

export CARLA_SERVER=${CARLA_ROOT}/CarlaUE4.sh
export PYTHONPATH=$PYTHONPATH:${CARLA_ROOT}/PythonAPI
export PYTHONPATH=$PYTHONPATH:${CARLA_ROOT}/PythonAPI/carla
export PYTHONPATH=$PYTHONPATH:$CARLA_ROOT/PythonAPI/carla/dist/carla-0.9.11-py3.7-win-amd64.egg
export SCENARIO_RUNNER_ROOT=${WORK_DIR}/scenario_runner
export LEADERBOARD_ROOT=${WORK_DIR}/leaderboard
export PYTHONPATH="${CARLA_ROOT}/PythonAPI/carla/":"${SCENARIO_RUNNER_ROOT}":"${LEADERBOARD_ROOT}":${PYTHONPATH}

export SCENARIOS=${WORK_DIR}/leaderboard/data/longest6/eval_scenarios.json
export ROUTES=${WORK_DIR}/leaderboard/data/longest6/longest6.xml
export REPETITIONS=1
export CHALLENGE_TRACK_CODENAME=SENSORS
export CHECKPOINT_ENDPOINT=${WORK_DIR}/results/transfuser_longest6.json
export TEAM_AGENT=${WORK_DIR}/team_code_transfuser/submission_agent.py
export TEAM_CONFIG=${WORK_DIR}/model_ckpt/transfuser
export DEBUG_CHALLENGE=0
export RESUME=1
export DATAGEN=0

python ${LEADERBOARD_ROOT}/leaderboard/leaderboard_evaluator_local.py \
--scenarios=${SCENARIOS}  \
--routes=${ROUTES} \
--repetitions=${REPETITIONS} \
--track=${CHALLENGE_TRACK_CODENAME} \
--checkpoint=${CHECKPOINT_ENDPOINT} \
--agent=${TEAM_AGENT} \
--agent-config=${TEAM_CONFIG} \
--debug=${DEBUG_CHALLENGE} \
--resume=${RESUME}

and my command is: 
(car) PS D:\Anaconda\envs\car\WindowsNoEditor\transfuser-2022\leaderboard\scripts> sh local_evaluation.sh D:\Anaconda\envs\car\WindowsNoEditor D:\Anaconda\envs\car\WindowsNoEditor\transfuser-2022
Could you please tell me if I have some problems with the content of my local_evaluation.sh file? Thanks a lot!

Kait0, 2024-11-26:
The first argument should be set to the CARLA folder. Did you actually call your CARLA folder WindowsNoEditor and stored it as an Anaconda env?

You could try commenting ` export PYTHONPATH=$PYTHONPATH:${CARLA_ROOT}/PythonAPI` can sometimes cause problems because there is a folder called carla in there which is not the carla package.

jacksonmygo, 2024-11-26:
Thank you a lot for your timely reply. I changed my CARLA simulator folder name from WindowsNoEditor to be CARLA_0.9.11 and commented export PYTHONPATH=$PYTHONPATH:${CARLA_ROOT}/PythonAPI. The anaconda env that I set for the project is called "car", but now I still have the same problem:
(car) PS D:\transfuser-2022\leaderboard\scripts> sh local_evaluation.sh D:\CARLA_0.9.11 D:\transfuser-2022
Traceback (most recent call last):
  File "D:\transfuser-2022/leaderboard/leaderboard/leaderboard_evaluator_local.py", line 466, in main
    leaderboard_evaluator = LeaderboardEvaluator(arguments, statistics_manager)
  File "D:\transfuser-2022/leaderboard/leaderboard/leaderboard_evaluator_local.py", line 87, in __init__
    dist = pkg_resources.get_distribution("carla")
  File "D:\Anaconda\envs\car\lib\site-packages\pkg_resources\__init__.py", line 471, in get_distribution
    dist = get_provider(dist)
  File "D:\Anaconda\envs\car\lib\site-packages\pkg_resources\__init__.py", line 347, in get_provider
    return working_set.find(moduleOrReq) or require(str(moduleOrReq))[0]
  File "D:\Anaconda\envs\car\lib\site-packages\pkg_resources\__init__.py", line 891, in require
    needed = self.resolve(parse_requirements(requirements))
  File "D:\Anaconda\envs\car\lib\site-packages\pkg_resources\__init__.py", line 777, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'carla' distribution was not found and is required by the application
Exception ignored in: <function LeaderboardEvaluator.__del__ at 0x00000218CCE6AE18>
Traceback (most recent call last):
  File "D:\transfuser-2022/leaderboard/leaderboard/leaderboard_evaluator_local.py", line 122, in __del__
    self._cleanup()
  File "D:\transfuser-2022/leaderboard/leaderboard/leaderboard_evaluator_local.py", line 134, in _cleanup
    if self.manager and self.manager.get_running_status() \
AttributeError: 'LeaderboardEvaluator' object has no attribute 'manager'
Traceback (most recent call last):
  File "D:\transfuser-2022/leaderboard/leaderboard/leaderboard_evaluator_local.py", line 476, in <module>
    main()
  File "D:\transfuser-2022/leaderboard/leaderboard/leaderboard_evaluator_local.py", line 472, in main
    del leaderboard_evaluator
UnboundLocalError: local variable 'leaderboard_evaluator' referenced before assignment
Could you please tell me how can I solve this problem? Thanks a lot!

Kait0, 2024-11-26:
Can you try and comment [these 4 lines](https://github.com/autonomousvision/transfuser/blob/3852e65e070f2efcfaba18fbbdb48057096b8a9e/leaderboard/leaderboard/leaderboard_evaluator_local.py#L87)?
and see if you get another error. The code has already [imported carla](https://github.com/autonomousvision/transfuser/blob/3852e65e070f2efcfaba18fbbdb48057096b8a9e/leaderboard/leaderboard/leaderboard_evaluator_local.py#L24) at this point so I am not sure why the pkg_manager can't find it.

jacksonmygo, 2024-11-27:
Thank you a lot for your reply! After I commented the four lines that you recommended, I got the problem shown below:
(car) PS D:\transfuser-2022\leaderboard\scripts> sh local_evaluation.sh D:\CARLA_0.9.11 D:\transfuser-2022
D:\Anaconda\envs\car\lib\site-packages\mmcv\__init__.py:21: UserWarning: On January 1, 2023, MMCV will release v2.0.0, in which it will remove components related to the training process and add a data transformation module. In addition, it will rename the package names mmcv to mmcv-lite and mmcv-full to mmcv. See https://github.com/open-mmlab/mmcv/blob/master/docs/en/compatibility.md for more details.
  'On January 1, 2023, MMCV will release v2.0.0, in which it will remove '

========= Preparing RouteScenario_0 (repetition 0) =========
> Setting up the agent
D:\transfuser-2022/model_ckpt/transfuser\model_seed1_39.pth
D:\transfuser-2022/model_ckpt/transfuser\model_seed2_39.pth
D:\transfuser-2022/model_ckpt/transfuser\model_seed3_37.pth
> Loading the world
Skipping scenario 'Scenario3' due to setup error: type object 'ParallelPolicy' has no attribute 'SUCCESS_ON_ONE'
Skipping scenario 'Scenario3' due to setup error: type object 'ParallelPolicy' has no attribute 'SUCCESS_ON_ONE'
Skipping scenario 'Scenario3' due to setup error: type object 'ParallelPolicy' has no attribute 'SUCCESS_ON_ONE'
Skipping scenario 'Scenario3' due to setup error: type object 'ParallelPolicy' has no attribute 'SUCCESS_ON_ONE'
Skipping scenario 'Scenario3' due to setup error: type object 'ParallelPolicy' has no attribute 'SUCCESS_ON_ONE'
Skipping scenario 'Scenario7' due to setup error: module 'py_trees.blackboard' has no attribute 'SetBlackboardVariable'
Skipping scenario 'Scenario9' due to setup error: module 'py_trees.blackboard' has no attribute 'SetBlackboardVariable'
Skipping scenario 'Scenario9' due to setup error: module 'py_trees.blackboard' has no attribute 'SetBlackboardVariable'
Skipping scenario 'Scenario7' due to setup error: module 'py_trees.blackboard' has no attribute 'SetBlackboardVariable'
Skipping scenario 'Scenario8' due to setup error: module 'py_trees.blackboard' has no attribute 'SetBlackboardVariable'
No more spawn points to use. Spawned 256 actors out of 500

The scenario could not be loaded:
> type object 'ParallelPolicy' has no attribute 'SUCCESS_ON_ONE'

Traceback (most recent call last):
  File "D:\transfuser-2022/leaderboard/leaderboard/leaderboard_evaluator_local.py", line 317, in _load_and_run_scenario
    scenario = RouteScenario(world=self.world, config=config, debug_mode=args.debug)
  File "D:\Anaconda\envs\car\lib\site-packages\leaderboard\scenarios\route_scenario_local.py", line 207, in __init__
    criteria_enable=criteria_enable)
  File "D:\Anaconda\envs\car\lib\site-packages\srunner\scenarios\basic_scenario.py", line 63, in __init__
    behavior = self._create_behavior()
  File "D:\Anaconda\envs\car\lib\site-packages\leaderboard\scenarios\route_scenario_local.py", line 492, in _create_behavior
    behavior = py_trees.composites.Parallel(policy=py_trees.common.ParallelPolicy.SUCCESS_ON_ONE)
AttributeError: type object 'ParallelPolicy' has no attribute 'SUCCESS_ON_ONE'

Could you please tell me how can I solve this problem? Thank you a lot!

Kait0, 2024-11-27:
You probably have the wrong version of pytrees installed (e.g. see [here](https://github.com/carla-simulator/scenario_runner/issues/23)).

jacksonmygo, 2024-11-29:
Thank you for your timely reply! After I turned on the visualization to get the output, I got the result shown below:
[image]
Could you please tell me if there are some problems with the output? and how can I solve the problem that the car just keeps stational? Thank you a lot!

Kait0, 2024-11-29:
The model is not loaded correctly. The green image should be semantic segmenation and the grey image depth.
You can debug [here](https://github.com/autonomousvision/transfuser/blob/3852e65e070f2efcfaba18fbbdb48057096b8a9e/team_code_transfuser/submission_agent.py#L89) what is going on. Probably the --agent-config= option is incorrectly set or there are other .pth files in the folder which are not the model.

jacksonmygo, 2024-11-30:
Thanks for your timely reply, and the model can start to work normally after I debug the "load model files" that you tell. I still have some problems with the results of the model, and I will start with a new issue. Thank you again for your patient guidance!

closed 2024-11-30 by jacksonmygo

jacksonmygo, 2025-01-15:
In addition, after I debug the code, I found where the lidar data gets lost:
[image]
As shown in figure, the lidar_transformed data gets lost exactly in line 552: "lidar_transformed = torch.from_numpy(lidar_to_histogram_features(lidar_transformed)).unsqueeze(0)", could you please tell me how to solve this problem? Thank you a lot!

jacksonmygo, 2025-01-15:
In addition, after I debug the code, I find where the lidar data gets lost, which is shown below:
[image]
In the figure, the lidar_transformed data becomes 0 exactly in line 552: "lidar_transformed = torch.from_numpy(lidar_to_histogram_features(lidar_transformed)).unsqueeze(0)". Could you please tell me how to solve this problem? Thank you a lot!