autonomousvision/transfuser #255: Some problems with the results of the local_evaluation.sh

jacksonmygo, 2024-11-30:
When I run the file local_evaluation.sh, I get some problems with the results, and the whole screenshots are shown below:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
Could you please tell me whether the Results of RouteScenario_0 are right or not? and how can I solve the errors happening in the Preparing RouteScenario_1 part? Thank you a lot!

Kait0, 2024-12-02:
The second error a sensor took too long is a usual crash where you just have to rerun the route. Maybe you can have a look [here](https://github.com/autonomousvision/transfuser/issues/216).

14 collisions on route 0 is too much though, so there is probably some problem.

jacksonmygo, 2024-12-05:
Thank you for your timely reply, and the second error "a sensor took too long" is solved after I rerun the route. For the problem that the number of collisions is too high which you mentioned, I reopen the visualization function and get the result shown below:
[image]
I guess that the problem is the LiDAR image can not load normally, but could you please tell me how can I solve this problem? Thank you a lot!

Kait0, 2024-12-05:
Yes that seems to be the problem. I don't know why it occurs. You need to debug [here](https://github.com/autonomousvision/transfuser/blob/3852e65e070f2efcfaba18fbbdb48057096b8a9e/team_code_transfuser/submission_agent.py#L211) and following to see where the LiDAR data gets lost.