autonomousvision/transfuser #256: How to test other backbone architectures besides the default one (Transfuser) such as (geometric fusion, late fusion...)

MCUBE-2023, 2024-12-02:
Dear author, 

I have a question regarding the changes that I should make in your code to test other architectures such as (geometric fusion, late fusion...) besides the default one which is Transfuser. To this end, I have the following questions:

1. For this end, do I have to set another option for the backbone **only** in the submsiion_agent.py (as illustrated in line 72 of screenshot 1 attached to this prompt) ? Or, in addition to this, I need **also** to change the value for the backbone in config.py (as illustrated in line 51of screenshot 2 attached to this prompt) ?
2. Is there any other changes that I need to make in these files or any other files for this end ? 

[image]
[image]

Kait0, 2024-12-03:
You don't have to make any changes, these configurations are automatically updated based on the args.txt file that come with the model (see [here](https://github.com/autonomousvision/transfuser/blob/3852e65e070f2efcfaba18fbbdb48057096b8a9e/team_code_transfuser/submission_agent.py#L72)).

MCUBE-2023, 2024-12-04:
So if I am understanding you correctly, in order to experiment the geometric fusion backbone for example, do I need to only (and I emphasize on the word 'only)'  change the content of the variable backbone located in config.py (line 51) from backbone = 'transfuser' to backbone = 'geometric_fusion' ?

Kait0, 2024-12-05:
No, you basically never need to change the default config in config.py.

For evaluation you only need to set the [--agent-config=](https://github.com/autonomousvision/transfuser/blob/3852e65e070f2efcfaba18fbbdb48057096b8a9e/leaderboard/scripts/local_evaluation.sh#L30C1-L30C16) to the folder with the geometric_fusion model.

If you want to train a new geometric_fusion model you set the [--backbone](https://github.com/autonomousvision/transfuser/blob/3852e65e070f2efcfaba18fbbdb48057096b8a9e/team_code_transfuser/train.py#L48) option in the training script to geometric_fusion.

MCUBE-2023, 2024-12-06:
Now, I understand you clearly. Thank you so much for your quick and clear reply.

closed 2024-12-10 by Kait0