autonomousvision/transfuser #257: Ask a question about training duration.

dd-xx-dot, 2024-12-05:
I am training on a single 4090 GPU with 24GB of VRAM, using the full 210GB dataset. When I set the batch size to 16, the VRAM usage is 14GB, and it takes 7 hours to complete one epoch, meaning the entire training will take about 12 days. However, when I increase the batch size to 24 or 32, VRAM usage increases to 23GB, but the training time per epoch becomes 9 or 10 hours, which is even longer. Could you help me understand why this is happening? I saw that in issue #110, a 3090 with 24GB of VRAM was used, and with a batch size of 32, the training only took 7.7 days

Kait0, 2024-12-05:
If you train without parallel (which is usually used for debugging) then parallel dataloading is turned off.
That can slow down training if you train with larger batch size because the data isn't loaded fast enough.
 You can turn it on [here](https://github.com/autonomousvision/transfuser/blob/3852e65e070f2efcfaba18fbbdb48057096b8a9e/team_code_transfuser/train.py#L162) by increasing the num_workers number.

dd-xx-dot, 2024-12-06:
Thank You so much for your response！！

closed 2024-12-06 by dd-xx-dot