autonomousvision/transfuser #260: The training loss did not converge.

dd-xx-dot, 2024-12-12:
I am using a 4090 GPU for single-card training, with a batch size of 32 and number_workers set to 24. The training was accidentally interrupted at the 16th epoch. I resumed training using the model output from the 15th epoch (model.sth) as the checkpoint file. However, the loss started increasing after resuming. What could be the reason for this? Here is the code I used for resuming training:
`conda activate tfusepython train.py --logdir /home/disk1/dx/transfuser/train_da
ta \
>                 --root_dir /home/disk1/dx/transfuser/data \
>                 --parallel_training 0 \
>                 --load_file /home/disk1/dx/transfuser/train_data/transfuser/model_15.pth \
> ^C
(tfuse) inspur@inspur-NF5280M5:/home/disk1/dx/transfuser/team_code_transfuser$ python train.py --logdir /home/disk1/dx/transfuser/train_data \
>                 --root_dir /home/disk1/dx/transfuser/data \
>                 --parallel_training 0 \
>                 --load_file /home/disk1/dx/transfuser/train_data/transfuser/model_15.pth \
>                 --start_epoch 16`
[image]

dd-xx-dot, 2024-12-13:
`/home/disk1/dx/transfuser/team_code_transfuser$ python train.py --logdir /home/disk1/dx/transfuser/train_data \                                       
> --root_dir /home/disk1/dx/transfuser/data \ [...]

dd-xx-dot, 2024-12-13:
`
/home/disk1/dx/transfuser/team_code_transfuser$ python train.py --logdir /home/disk1/dx/transfuser/train_data \                                       
> --root_dir /home/disk1/dx/transfuser/data \ [...]
`

dd-xx-dot, 2024-12-13:
Sorry, this is a GitHub code display error; it should be as follows：
/home/disk1/dx/transfuser/team_code_transfuser$ python train.py --logdir /home/disk1/dx/transfuser/train_data \                                       
> --root_dir /home/disk1/dx/transfuser/data \ [...]

dd-xx-dot, 2024-12-13:
[image]

dd-xx-dot, 2024-12-13:
During checkpoint training, the following issue occurred:
File "train.py", line 399, in <module>
    main()
  File "/home/inspur/anaconda3/envs/tfuse/lib/python3.7/site-packages/torch/distributed/elastic/multiprocessing/errors/__init__.py", line 345, in wrapper
    return f(*args, **kwargs)
  File "train.py", line 200, in main
    trainer.train()
  File "train.py", line 314, in train
    self.optimizer.step()
  File "/home/inspur/anaconda3/envs/tfuse/lib/python3.7/site-packages/torch/optim/optimizer.py", line 109, in wrapper
    return func(*args, **kwargs)
  File "/home/inspur/anaconda3/envs/tfuse/lib/python3.7/site-packages/torch/autograd/grad_mode.py", line 27, in decorate_context
    return func(*args, **kwargs)
  File "/home/inspur/anaconda3/envs/tfuse/lib/python3.7/site-packages/torch/optim/adamw.py", line 175, in step
    capturable=group['capturable'])          
  File "/home/inspur/anaconda3/envs/tfuse/lib/python3.7/site-packages/torch/optim/adamw.py", line 231, in adamw
    capturable=capturable)                   
  File "/home/inspur/anaconda3/envs/tfuse/lib/python3.7/site-packages/torch/optim/adamw.py", line 259, in _single_tensor_adamw
    assert not step_t.is_cuda, "If capturable=False, state_steps should not be CUDA tensors."
AssertionError: If capturable=False, state_steps should not be CUDA tensors.
/To address this issue, I added the following code at line 185 in train.py:
for param_group in optimizer.param_groups:
    param_group['capturable'] = True
I am unsure if this will have any impact.

Kait0, 2024-12-14:
as far as I know you do not need to change the code to restart the training. Did you change the pytorch version?

st3lzer, 2024-12-14:
I also encountered this issue with capturable=False and needed to solve it by adding

```
for param_group in optimizer.param_groups:
    param_group['capturable'] = True
```

But this did not alter the outcome of the training in any way.

dd-xx-dot, 2024-12-18:
I didn't change the PyTorch version and retrained the model from scratch, which resolved the issue. Thank you for your work and timely help.
I would like to ask two more questions. Could you please help me with these?
1、The evaluation is very slow, and it has already taken three days. Are there any ways to accelerate the process?
2、I am training and evaluating on a remote server and cannot see the running process in CARLA. Are there any methods for visualizing the process?"

dd-xx-dot, 2024-12-18:
> I also encountered this issue with capturable=False and needed to solve it by adding
> 
> ```
> for param_group in optimizer.param_groups:
>     param_group['capturable'] = True
> ```
> 
> But this did not alter the outcome of the training in any way.

Thank you! I will try to find if there are any other reasons for the issue.

Kait0, 2024-12-18:
1. Yes, using multiple GPUs, see for example [here](https://github.com/autonomousvision/carla_garage/tree/leaderboard_1?tab=readme-ov-file#how-to-actually-evaluate), or using multiple servers on a big GPU.
2. Yes, you can use the visualization code by setting the environment variable [SAVE_PATH](https://github.com/autonomousvision/transfuser/blob/3852e65e070f2efcfaba18fbbdb48057096b8a9e/team_code_transfuser/submission_agent.py#L23) and [debug=True](https://github.com/autonomousvision/transfuser/blob/3852e65e070f2efcfaba18fbbdb48057096b8a9e/team_code_transfuser/config.py#L35C5-L35C11).

dd-xx-dot, 2024-12-18:
Thank you once again for your help.

dd-xx-dot, 2024-12-18:
The visualized images, when the image sequence number is a multiple of 10, suddenly include some unrelated images that are not part of the running path.

Here are a few examples, and the numbering seems strange as well.
399
[image]
4
[image]
40
[image]
400
[image]

409
[image]
41
[image]
410
[image]

42
[image]

dd-xx-dot, 2024-12-18:
> > The evaluation results I obtained are relatively poor, and I used a single model from the 41st epoch for the evaluation. Would using the model from the 31st epoch yield better results? Below are my evaluation scores: Avg. driving score,38.002 Avg. route completion,79.633 Avg. infraction penalty,0.478 Collisions with pedestrians,0.09157889340023193 Collisions with vehicles,1.9918409314550445 Collisions with layout,0.2518419568506378 Red lights infractions,0.1602630634504059 Stop sign infractions,0.3434208502508697 Off-road infractions,0.01211817706918569 Route deviations,0.0 Route timeouts,0.09157889340023193 Agent blocked,0.3663155736009277
> 
> I think the problem might be that you used only a single model and the publication uses an ensemble of three.

ok,thanks,i will try again with an ensemble of three

Kait0, 2024-12-18:
Your editor just orders the images incorrectly because they don't have leading zeros. 4 does not come after 399 it comes after 3.
You should also use epoch 31. It is usually the best or similar to the best.
You can check the issues for other people with similar problems.

dd-xx-dot, 2025-01-11:
I still can't achieve the scores reported in the original paper, even when I use models 31, 33, and 37 together for evaluation, or evaluate using model 31 alone. In fact, the scores from these two evaluations are even lower than the scores I obtained during my first evaluation using model 41.

Below are the scores for model 41, models 31, 33, and 37, as well as model 31 individually.
model41：
Avg. driving score,38.002
Avg. route completion,79.633
Avg. infraction penalty,0.478
Collisions with pedestrians,0.09157889340023193
Collisions with vehicles,1.9918409314550445
Collisions with layout,0.2518419568506378
Red lights infractions,0.1602630634504059
Stop sign infractions,0.3434208502508697
Off-road infractions,0.01211817706918569
Route deviations,0.0
Route timeouts,0.09157889340023193
Agent blocked,0.3663155736009277

model1, 33, and 37:
Avg. driving score,35.188
Avg. route completion,78.13
Avg. infraction penalty,0.4650000000000001
Collisions with pedestrians,0.02356670883595531
Collisions with vehicles,2.050303668728112
Collisions with layout,0.3535006325393296
Red lights infractions,0.11783354417977654
Stop sign infractions,0.2592337971955084
Off-road infractions,0.09677668983485048
Route deviations,0.0
Route timeouts,0.14140025301573186
Agent blocked,0.3299339237033743

model31:
Avg. driving score,25.186
Avg. route completion,74.616
Avg. infraction penalty,0.397
Collisions with pedestrians,0.07198298448124002
Collisions with vehicles,2.759347738447534
Collisions with layout,0.5038808913686801
Red lights infractions,0.1919546252833067
Stop sign infractions,0.4318979068874401
Off-road infractions,0.12354199683233621
Route deviations,0.0
Route timeouts,0.16796029712289337
Agent blocked,0.33592059424578674

Kait0, 2025-01-14:
Do you also get these scores with the pre-trained models provided in the repository?
That way you can exclude that there is some issue with your training.

Your results seem to have an increased blocked value resulting in a lower route completion.
I think there were a few users in the past who had this type of issue (you can check the github issues, if there is something helpful).
Ultimately I don't think we figured out were this comes from in these cases
It could be that the different hardware produces different renderings compared to the dataset (this is a known problem of 30XX GPUs) and that the model is not robust to this distribution shift. If you get the same result with the pre-trained models than that might be the case.

dd-xx-dot, 2025-01-15:
The code comment here seems to have an issue. In the model.py, the comment on [line 290 ](https://github.com/autonomousvision/transfuser/blame/3852e65e070f2efcfaba18fbbdb48057096b8a9e/team_code_transfuser/model.py#L290)indicates that the format of gt_bboxes is [top-left x-coordinate of the bounding box, top-left y-coordinate of the bounding box, bottom-right x-coordinate of the bounding box, bottom-right y-coordinate of the bounding box]. However, based on the code below, I infer that the format of gt_bboxes should be [center x-coordinate, center y-coordinate, bounding box width, bounding box height]. The code is as follows:
```
center_x = gt_bbox[:, [0]] * width_ratio
center_y = gt_bbox[:, [1]] * width_ratio
gt_centers = torch.cat((center_x, center_y), dim=1)

scale_box_h = gt_bbox[j, 3] * height_ratio
scale_box_w = gt_bbox[j, 2] * width_ratio```

Kait0, 2025-01-15:
The coordinate systems in this repository are a big mess and change throughout the code. I would not trust any comment.
I fixed this in our followup repo [carla_garage](https://github.com/autonomousvision/carla_garage/) but that was a lot of effort, I won't change things here.

closed 2025-01-23 by dd-xx-dot