autonomousvision/transfuser #266: The lidar BEV can not load normally in the visualization, but the lidar data does not lose in submission_agent.

jacksonmygo, 2025-01-14:
I open the visualization function and get the result shown below:
[image]
The lidar BEV can not load normally. However, I print out the lidar data in submission_agent, and I get the result:
[image]
Could you please tell me which part is wrong, and how can I solve this problem? Thank you a lot!

Kait0, 2025-01-14:
Hm hard to tell, did you change something about the code?
Could be some coordinate system issue, or have something to do that the LiDAR is only available every second frame (due to being 10 Hz) which is why we use action repeat 2.

jacksonmygo, 2025-01-15:
Thank you for your timely reply! I changed nothing about the code. I guess it is because what you mentioned that "the LiDAR is only available every second frame", which is shown below:
[image]
You can see that for tick_data, it appears twice before lidar_bev appears once. So you mentioned you use action repeat 2 to solve it, but I do not understand what it means, could you please tell me where the "action repeat 2" code is? Thank you a lot!

jacksonmygo, 2025-01-15:
In addition, I found where the data gets wrong, which is shown below:
[image]
[image]
It is in the file data.py, and I found that the data gets wrong exactly in line 460: "hist = np.histogramdd(point_cloud[..., :2], bins=(xbins, ybins))[0]" in the figure, could you please tell me how can I solve this problem? Thank you a lot!

Kait0, 2025-01-15:
Yes the second tick with empty LiDAR is still computed (for GPS) but not used.
The run_step function returns in this case without using the data:
See [here](https://github.com/autonomousvision/transfuser/blob/3852e65e070f2efcfaba18fbbdb48057096b8a9e/team_code_transfuser/submission_agent.py#L252) and [here](https://github.com/autonomousvision/transfuser/blob/3852e65e070f2efcfaba18fbbdb48057096b8a9e/team_code_transfuser/config.py#L156).

There probably is no problem to solve. The function you mention just turns the point cloud to the image. Since the BEV image only is in front of the car it will be empty when the half LiDAR sweep you get is from the back of the car.

jacksonmygo, 2025-01-16:
Thank you for your timely reply! I try to only output the "hist" data, but it seems that it hardly outputs any data, and the visualization output shows that there is actually no clear lidar cloud image, which is shown below:
[image]
[image]
However, I also find that for some images, there are a few white points at the bottom in nearly middle of the image, which is shown below:
[image]
[image]
I do not know whether the lidar cloud BEV image is loaded or not, but it seems can not appear normally, so I think there is still some problem. Could you please tell me how to solve it? Thank you a lot!

Kait0, 2025-01-17:
hm don't know this should not happen if you have not changed the code.
Can have various reasons, the index being messed up (and only the back lidar being visualized), the coordinate system of the point cloud might have changed, the lidar was modified without deepcopy etc.
Do you use the build in visualization code with debug=True or call the function by yourself?

jacksonmygo, 2025-01-18:
Yes, I set the visualization code with debug=True, and I remember that what the code I only change is shown below:
[image]
I comment these 4 lines because I followed your instruction in #254 , but I think this should not affect the visualization result. This really makes me very confused. I guess it may be caused by the change of the coordinate system of the point cloud. Could you please tell me where all of the coordinate system codes are? Thank you a lot!

jacksonmygo, 2025-01-20:
Yes, I set the visualization code with debug=True, and now I find where the problem is. I output the value of point_cloud data, which is shown below:
[image]
[image]
You can see that the y-axis value of point_cloud data is always positive, but in the code in data.py, the y-axis value should be always negative, which is shown below:
[image]
As a result, I think that the problem is what you mentioned "only the back lidar being visualized", and I used visualization function and confirmed my guess. Could you please tell me how to solve this problem? Thank you a lot!

Kait0, 2025-01-20:
[image]

Just downloaded the code again and run it on my desktop. I can't reproduce this issue, I can see the LiDAR.
Perhaps you made some change to the code and forgot about it? Maybe you can try downloading the code again.

jacksonmygo, 2025-01-20:
Thank you for your reply. I download the code again, and run it on my desktop, but the front lidar point cloud still cannot load, with the result shown below:
[image]
And I added the code that "point_cloud[..., 1] = - point_cloud[..., 1]" to make the y-axis data to be all negative, and the result is shown below:
[image]
You can see that the back lidar point can load correctly, so I think the reason is that only the back lidar works and the front lidar does not work. Could you please tell me where the codes for front lidar and back lidar working? Thank you a lot!

Kait0, 2025-01-20:
The LiDAR sensor alternates between generating a sweep for the front and one for the back.
The code should only proceed to the visualization function when the front frame is available, for some reason the opposite seems to happen for you.
Maybe something wrong with the action repeat?
You can try to comment [this](https://github.com/autonomousvision/transfuser/blob/3852e65e070f2efcfaba18fbbdb48057096b8a9e/team_code_transfuser/submission_agent.py#L252) then it should alternate between front and back.

jacksonmygo, 2025-01-20:
Yes, it is this problem! Now I comment that code and the visualization now can work well. Thank you again for your patient guidance!

closed 2025-01-20 by jacksonmygo

Kait0, 2025-01-21:
Great. Just make sure you don't input the frames with no LiDAR into the model, that might reduce performance.
Not sure why this happens in your case but maybe you just need to add +1 to the [step](https://github.com/autonomousvision/transfuser/blob/3852e65e070f2efcfaba18fbbdb48057096b8a9e/team_code_transfuser/submission_agent.py#L38C9-L38C23) counter to switch between even/odd frames.