autonomousvision/transfuser #268: PyTorch and CUDA version

CoderXuans, 2025-01-17:
Dear author,
I ran the `./leaderboard/scripts/local_evaluation.sh` script and encountered the following error. I installed the project-recommended `torch-1.11.0+cu102`.
The error is as follows：
NVIDIA GeForce RTX 4090 with CUDA capability sm_89 is not compatible with the current PyTorch instal

CoderXuans, 2025-01-17:
@Kait0 
Dear author,
I'm currently downloading torch1.13.1+cu117 because I've found that the highest version supported by MMCV is cuda 11.7+torch1.13.x. The version of mmcv I'm downloading is mmcv==1.7.1.It is ok?

Kait0, 2025-01-17:
I don't know which version combinations work, you need to check the documentation of mmcv etc.

CoderXuans, 2025-01-17:
ok，thanks

closed 2025-01-17 by Kait0