autonomousvision/transfuser #270: Carla version mismatch

jacksonmygo, 2025-01-19:
When I use the version CARLA_0.9.10.1, the local_evaluation.sh cannot work properly, with the wrong warnings shown below:
(car) D:\transfuser-2022\leaderboard\scripts>sh local_evaluation.sh D:\CARLA_0.9.10.1 D:\transfuser-2022
WARNING: Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API
WARNING: Client API version     = 0.9.11
WARNING: Simulator API version  = 0.9.10.1
Traceback (most recent call last):
  File "D:\transfuser-2022/leaderboard/leaderboard/leaderboard_evaluator_local.py", line 466, in main
    leaderboard_evaluator = LeaderboardEvaluator(arguments, statistics_manager)
  File "D:\transfuser-2022/leaderboard/leaderboard/leaderboard_evaluator_local.py", line 84, in __init__
    self.world = self.client.load_world('Town01')
RuntimeError: rpc::rpc_error during call in function load_new_episode
Exception ignored in: <function LeaderboardEvaluator.__del__ at 0x000001FBEC0CC0D0>
Traceback (most recent call last):
  File "D:\transfuser-2022/leaderboard/leaderboard/leaderboard_evaluator_local.py", line 122, in __del__
    self._cleanup()
  File "D:\transfuser-2022/leaderboard/leaderboard/leaderboard_evaluator_local.py", line 134, in _cleanup
    if self.manager and self.manager.get_running_status() \
AttributeError: 'LeaderboardEvaluator' object has no attribute 'manager'
Traceback (most recent call last):
  File "D:\transfuser-2022/leaderboard/leaderboard/leaderboard_evaluator_local.py", line 476, in <module>
    main()
  File "D:\transfuser-2022/leaderboard/leaderboard/leaderboard_evaluator_local.py", line 472, in main
    del leaderboard_evaluator
UnboundLocalError: local variable 'leaderboard_evaluator' referenced before assignment
Could you please tell me how can I solve this problem? Thank you a lot!

jacksonmygo, 2025-01-19:
I do not know why the client API version is 0.9.11, could you please tell me where can I know the client API version? Thank you.

Kait0, 2025-01-20:
Depends on how you added the client.
If it is easy_install/ pip installed then you can check with pip list which CARLA version is installed and uninstall the wrong one.
Other option is you added the .egg file to the python path. In that case you need to remove the wrong .egg file from the PYTHONPATH environment variable.

jacksonmygo, 2025-01-20:
Here is the code for the local_evaluation.sh file:
export CARLA_ROOT=${1:-/home/kchitta/Documents/CARLA_0.9.10}
export WORK_DIR=${2:-/home/kchitta/Documents/transfuser}

export CARLA_SERVER=${CARLA_ROOT}/CarlaUE4.sh
# export PYTHONPATH=$PYTHONPATH:${CARLA_ROOT}/PythonAPI
export PYTHONPATH=$PYTHONPATH:${CARLA_ROOT}/PythonAPI/carla
export PYTHONPATH=$PYTHONPATH:$CARLA_ROOT/PythonAPI/carla/dist/carla-0.9.10-py3.7-win-amd64.egg
export SCENARIO_RUNNER_ROOT=${WORK_DIR}/scenario_runner
export LEADERBOARD_ROOT=${WORK_DIR}/leaderboard
export PYTHONPATH="${CARLA_ROOT}/PythonAPI/carla/":"${SCENARIO_RUNNER_ROOT}":"${LEADERBOARD_ROOT}":${PYTHONPATH}
export SAVE_PATH=${WORK_DIR}/visualization

export SCENARIOS=${WORK_DIR}/leaderboard/data/longest6/eval_scenarios.json
export ROUTES=${WORK_DIR}/leaderboard/data/longest6/longest6.xml
export REPETITIONS=1
export CHALLENGE_TRACK_CODENAME=SENSORS
export CHECKPOINT_ENDPOINT=${WORK_DIR}/results/transfuser_longest6.json
export TEAM_AGENT=${WORK_DIR}/team_code_transfuser/submission_agent.py
export TEAM_CONFIG=${WORK_DIR}/model_ckpt/transfuser
export DEBUG_CHALLENGE=0
export RESUME=1
export DATAGEN=0

python ${LEADERBOARD_ROOT}/leaderboard/leaderboard_evaluator_local.py \
--scenarios=${SCENARIOS}  \
--routes=${ROUTES} \
--repetitions=${REPETITIONS} \
--track=${CHALLENGE_TRACK_CODENAME} \
--checkpoint=${CHECKPOINT_ENDPOINT} \
--agent=${TEAM_AGENT} \
--agent-config=${TEAM_CONFIG} \
--debug=${DEBUG_CHALLENGE} \
--resume=${RESUME}
You can see that I set the PYTHONPATH to carla 0.9.10 .egg file, but it still tells that the Client API version = 0.9.11. Could you please tell me how can I solve this problem? Thank you a lot!

jacksonmygo, 2025-01-20:
Here is the code for the local_evaluation.sh file:
export CARLA_ROOT=${1:-/home/kchitta/Documents/CARLA_0.9.10}
export WORK_DIR=${2:-/home/kchitta/Documents/transfuser}

export CARLA_SERVER=${CARLA_ROOT}/CarlaUE4.sh
#export PYTHONPATH=$PYTHONPATH:${CARLA_ROOT}/PythonAPI
export PYTHONPATH=$PYTHONPATH:${CARLA_ROOT}/PythonAPI/carla
export PYTHONPATH=$PYTHONPATH:$CARLA_ROOT/PythonAPI/carla/dist/carla-0.9.10-py3.7-win-amd64.egg
export SCENARIO_RUNNER_ROOT=${WORK_DIR}/scenario_runner
export LEADERBOARD_ROOT=${WORK_DIR}/leaderboard
export PYTHONPATH="${CARLA_ROOT}/PythonAPI/carla/":"${SCENARIO_RUNNER_ROOT}":"${LEADERBOARD_ROOT}":${PYTHONPATH}
export SAVE_PATH=${WORK_DIR}/visualization

export SCENARIOS=${WORK_DIR}/leaderboard/data/longest6/eval_scenarios.json
export ROUTES=${WORK_DIR}/leaderboard/data/longest6/longest6.xml
export REPETITIONS=1
export CHALLENGE_TRACK_CODENAME=SENSORS
export CHECKPOINT_ENDPOINT=${WORK_DIR}/results/transfuser_longest6.json
export TEAM_AGENT=${WORK_DIR}/team_code_transfuser/submission_agent.py
export TEAM_CONFIG=${WORK_DIR}/model_ckpt/transfuser
export DEBUG_CHALLENGE=0
export RESUME=1
export DATAGEN=0

python ${LEADERBOARD_ROOT}/leaderboard/leaderboard_evaluator_local.py \
--scenarios=${SCENARIOS}  \
--routes=${ROUTES} \
--repetitions=${REPETITIONS} \
--track=${CHALLENGE_TRACK_CODENAME} \
--checkpoint=${CHECKPOINT_ENDPOINT} \
--agent=${TEAM_AGENT} \
--agent-config=${TEAM_CONFIG} \
--debug=${DEBUG_CHALLENGE} \
--resume=${RESUME}
You can see that I set the PYTHONPATH to carla 0.9.10 .egg file, but it still tells that the Client API version = 0.9.11. Could you please tell me how can I solve this problem? Thank you a lot!

Kait0, 2025-01-25:
maybe there are multiple carla paths in the python path. You can try printing the environment variable.
If carla says its 0.9.11 then it must be installed somewhere on your system. You can try moving it to a different place.

closed 2025-02-04 by Kait0