autonomousvision/transfuser #276: preds, list index out of range

amanysh99, 2025-02-11:
I'm encountering an issue while training the model. Specifically, I'm running into an " list index out of range" error with the preds list as attachment "out of index" indicates

[image]

. To troubleshoot this, I added a try-except block to print a warning message, which revealed that the preds list has fewer than 7 elements.   And that is in the line: https://github.com/autonomousvision/transfuser/blob/3852e65e070f2efcfaba18fbbdb48057096b8a9e/team_code_transfuser/model.py#L775

[image]

amanysh99, 2025-02-11:
What's puzzling is that the line of code that uses also preds: https://github.com/autonomousvision/transfuser/blob/3852e65e070f2efcfaba18fbbdb48057096b8a9e/team_code_transfuser/model.py#L709 doesn't seem to throw this error. I'm unsure why this discrepancy exists.

Kait0, 2025-02-11:
I would recommend that you use a debugger and step through the code line by line to see what is going on.
Perhaps some problem with the [multi-apply function](https://github.com/autonomousvision/transfuser/blob/3852e65e070f2efcfaba18fbbdb48057096b8a9e/team_code_transfuser/model.py#L125C16-L125C27).
Are you using a different mmdet version?
forward_single should always return 7 elements.

amanysh99, 2025-02-11:
Thank you, I've resolved the issue with the multi_apply function, but now the process is being killed without any error message!

[image]

Kait0, 2025-02-11:
Killed can mean that linux shut down the process because it was exceeding RAM. You can try monitoring RAM usage.

amanysh99, 2025-02-11:
After monitoring the RAM usage, I found that RAM has exceeded its limit. 
Increasing the RAM to 32 GB might solve the problem. Let me know if this is possible?

[image]

Kait0, 2025-02-11:
I don't know if 32 GB is enough we typically use more since we run training on a server with multiple GPUs.
If you just want to debug locally you can try only loading a subset of the dataset.

closed 2025-02-16 by Kait0