autonomousvision/transfuser #277: How to evaluate the specified scenario in the CARLA for the model?

jacksonmygo, 2025-02-16:
There are more than 30 scenarios in the test to evaluate the agent in CARLA. However, each time when I reopen the CARLA and the computer, it will begin back to RouteScenario_0 to evaluate. Could you please tell me if there is some place can I adjust so that I can evaluate the specified scenario in the CARLA for the model, such as RouteScenario_3, RouteScenario_16, and so on? Thank you a lot!

Kait0, 2025-02-16:
You can find the individual routes [here](https://github.com/autonomousvision/transfuser/tree/2022/leaderboard/data/longest6/longest6_split).

jacksonmygo, 2025-02-17:
[image]
Thanks for your timely reply. Could you please tell me what does it mean? Do you mean that if I want to evaluate the specified scenario, for example RouteScenario_16, I need to take the longest_weather_16.xml file at the first line of the folder? Thank you.

Kait0, 2025-02-17:
yes you if you set the --routes= option to longest_weather_16.xml it will evaluate route 16.
Note that scenarios in CARLA mean specific safety critical situations within a route.
You usually set --scenarios= to ${WORK_DIR}/leaderboard/data/longest6/eval_scenarios.json which contains the scenario definitions for all routes.

closed 2025-02-19 by jacksonmygo