autonomousvision/transfuser #278: can't find agents.navigation.global_route_planner_dao

jamesvilho, 2025-02-22:
I am trying to generate a dataset, however, i am getting an error that global_route_planner_dao is not found in the repo. The agents module (which should contain navigation.global_route_planner_dao) does not exists. 

[image]

Below is datagen.sh file from ./leaderboard/scripts/datagen.sh /home/uwb/CARLA_0.9.15 /home/uwb/Desktop/transfuser-2022 that I am trying to execute. Is there anything I need to change in this datagen.sh file? Thanks 

[image]

Kait0, 2025-02-22:
> Below is datagen.sh file from ./leaderboard/scripts/datagen.sh /home/uwb/CARLA_0.9.15 

You seem to be using the wrong CARLA version, you need 0.9.10.1 for this repo

jamesvilho, 2025-02-22:
It's okay. Instead of generating my dataset, I managed to download/use your dataset.

closed 2025-02-24 by Kait0