autonomousvision/transfuser #280: Infinite loop while loading the lidar folders!

amanysh99, 2025-02-27:
During the execution of the code, after switching to CPU processing and selecting a subset of the dataset (specifically, the 'rr_dataset_23_11' folder), I encountered an issue where the program appears to enter an infinite loop while attempting to load 4545 lidar files from three folders. I am currently unable to determine the cause of this issue.

[image]

Kait0, 2025-02-27:
Can you debug and report in which line of code the training enters an infinite loop?
Have not seen this before?

amanysh99, 2025-02-27:
Umm while debugging the code it enters an infinite loop here:

[image]

amanysh99, 2025-02-27:
Here is the train.py and model.py after switching to CPU: https://github.com/amanysh99/Carla

Kait0, 2025-02-27:
hm? the main function is not very explicit.

maybe its the multiprocessing strategy.
You could try changing the fork method to 
mp.set_start_method('spawn')
or
mp.set_start_method('forkserver')

fork is the default but can sometimes deadlock. You can try the other two and see if it helps.

amanysh99, 2025-02-28:
unfortunately i had tried the two option sorrowfully all in veins :(

Kait0, 2025-02-28:
https://github.com/autonomousvision/transfuser/blob/3852e65e070f2efcfaba18fbbdb48057096b8a9e/team_code_transfuser/train.py#L159
I would change the num workers in the dataloaders to 0.
That will turn off parallelism and helps you debug easier.
You can try debugging then again, you should get a better error message of where the code enters an infinite loop.

amanysh99, 2025-02-28:
It enters an infinite loop again without displaying any obvious error message!

Kait0, 2025-02-28:
and you can't debug?
If debugging doesn't work you can go the traditional way with placing print( statements everywhere to see where the code hangs.

amanysh99, 2025-03-01:
While debugging to check that the root directory path is found:
import os
root_dir = r'D:/Lab 6/transfuser/team_code_transfuser/data/rr_dataset_23_11'
print("Path exists:", os.path.exists(root_dir))  # Should print "True"
print("Subdirectories:", os.listdir(root_dir))  # Should list dataset subfolders (e.g., Town01)

 It outputs that : FileNotFoundError: [Errno 2] No such file or directory: 'D:/Lab 6/transfuser/team_code_transfuser/data/rr_dataset_23_11'

Kait0, 2025-03-01:
aren't you mixing windows style paths with Linux style paths?

Also I would avoid putting spaces in paths, should generally work but sometimes not.

r'D:\Lab 6\transfuser\team_code_transfuser\data\rr_dataset_23_11'
should work python can handle windows paths.
Does the folder actually exist if you post this path in the explorer?

amanysh99, 2025-03-01:
Yes that exactly my fault that I mixed Windows and Linux style paths, but I solved and after checking the folder exist, but still enters an infinite loop:(

closed 2025-03-05 by amanysh99