autonomousvision/transfuser #281: Question about the result in the paper

jacksonmygo, 2025-03-05:
When I read the result of the paper "TransFuser Imitation with Transformer-Based Sensor Fusion for Autonomous Driving", I am a little confused with the representations of the results:
[image]
Could you please tell me why the DS and RC have minus or plus signal following each number, like this:
[image]
Does it should be a certain number? In addition, I want to know what does the "Block: Vehicle Blocked" in the result part mean? Thank you a lot!

jacksonmygo, 2025-03-08:
Thank you for your timely reply! In addition, I want to know that where is the code for setting the speed of the autonomous vehicle? Thank you a lot!

Kait0, 2025-03-10:
You don't set the speed you set, brake throttle and steer in carla, see [here](https://github.com/autonomousvision/transfuser/blob/3852e65e070f2efcfaba18fbbdb48057096b8a9e/team_code_transfuser/submission_agent.py#L376).

closed 2025-03-12 by Kait0