autonomousvision/transfuser #285: Problems of download dataset

Ran12256, 2025-03-10:
I met the problem when downloading the dataset when running download_data.sh:

Connecting to s3.eu-central-1.amazonaws.com (s3.eu-central-1.amazonaws.com)|108.160.165.11|:443... failed: Connection timed out.

How to solve this problem? How to download the dataset?

Ran12256, 2025-03-10:
[image]
I tried to visit the webset but it didn't work.

Kait0, 2025-03-10:
Yes the 2022 code had quite a lot of changes / updates compared to the 2021 code.
You can have a look a the [data.py](https://github.com/autonomousvision/transfuser/blob/2022/team_code_transfuser/data.py) file of the 2022 branch to see how it is loaded there.

Ran12256, 2025-03-10:
> You need to check if you can access for example this link:
> 
> https://s3.eu-central-1.amazonaws.com/avg-projects/transfuser/data/2022_data/left.zip
> 
> There are 13 files see [here](https://github.com/autonomousvision/transfuser/blob/3852e65e070f2efcfaba18fbbdb48057096b8a9e/download_data.sh#L10).
> 
> If you are located in China you need a (high speed) VPN to download the data since its located in EU.

Thank you! It works.

closed 2025-03-10 by Ran12256

Captain6606, 2025-05-25:
> 您需要检查是否可以访问，例如此链接：
> 
> https://s3.eu-central-1.amazonaws.com/avg-projects/transfuser/data/2022_data/left.zip
> 
> 这里有 13 个文件，请看[这里](https://github.com/autonomousvision/transfuser/blob/3852e65e070f2efcfaba18fbbdb48057096b8a9e/download_data.sh#L10)。
> 
> 如果您位于中国，则需要（高速）VPN 来下载数据，因为它位于欧盟。

Dear author, I also encountered a similar problem. After downloading these data to my local machine, I should transfer them to the specific folder on the server. Looking forward to your reply. Thank you very much.

Captain6606, 2025-05-25:
> 您需要检查是否可以访问，例如此链接：
> 
> https://s3.eu-central-1.amazonaws.com/avg-projects/transfuser/data/2022_data/left.zip
> 
> 这里有 13 个文件，请看[这里](https://github.com/autonomousvision/transfuser/blob/3852e65e070f2efcfaba18fbbdb48057096b8a9e/download_data.sh#L10)。
> 
> 如果您位于中国，则需要（高速）VPN 来下载数据，因为它位于欧盟。

Dear author, I have also encountered a similar problem. After downloading these data to my local machine, where should I transfer them to in the server's folder? Looking forward to your reply. Thank you very much.

Kait0, 2025-05-26:
I do not quite understand what the problem is that you are encountering. The data is not available on Baidu, it is stored on a AWS server. You might need a VPN to download it from China.

Captain6606, 2025-05-27:
> I do not quite understand what the problem is that you are encountering. The data is not available on Baidu, it is stored on a AWS server. You might need a VPN to download it from China.

Thank you very much for your reply. I have solved the problem.