autonomousvision/transfuser #286: Some issues during the evaluation

dd-xx-dot, 2025-03-12:
`(base) ubuntu@ubuntu:~/dx/transfuser$ ./leaderboard/scripts/local_evaluation.sh
/home/ubuntu/dx/transfuser/leaderboard/leaderboard/leaderboard_evaluator_local.py:22: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
./leaderboard/scripts/local_evaluation.sh：行 37: 25298 段错误               （核心已转储） python3 ${LEADERBOARD_ROOT}/leaderboard/leaderboard_evaluator_local.py --scenarios=${SCENARIOS} --routes=${ROUTES} --repetitions=${REPETack=${CHALLENGE_TRACK_CODENAME} --checkpoint=${CHECKPOINT_ENDPOINT} --agent=${TEAM_AGENT} --agent-config=${TEAM_CONFIG} --debug=${DEBUG_CHALLENGE} --resume=${RESUME}
`

dd-xx-dot, 2025-03-12:
1、I would like to ask, why does the score get lower as the rain gets lighter? Is it because the routes are different?

[image]

2、Is it possible to specify the weather conditions for each route? And can the randomly generated scenario agent be saved for subsequent evaluations? I noticed in your paper that it is generated randomly every time.

Kait0, 2025-03-12:
I don't know what you are evaluating here but you should have a look at the standard deviations which are > 30 in most cases.
This means the evaluation was not done with enough seeds / routes. Which such a high std all the results are basically random (assuming normal distribution and 2 sigma rule you have a confidence interval of +- 60 DS here).

> ./leaderboard/scripts/local_evaluation.sh：行 37: 25298 段错误               （核心已转储） python3 ${LEADERBOARD_ROOT}/leaderboard/leaderboard_evaluator_local.py --scenarios=${SCENARIOS} --routes=${ROUTES} [...]

The chinese characters translate Segment error (core dumped), which suggests that there is a bug that should not happen.

2. The weather is specified alongside the [route](https://github.com/autonomousvision/transfuser/blob/3852e65e070f2efcfaba18fbbdb48057096b8a9e/leaderboard/data/longest6/longest6.xml#L25)

I do not know what you mean by " randomly generated scenario agent". Could you elaborate?

Kait0, 2025-03-13:
Your csv looks fine. You only evaluated one repetition of longest6, so the weather, daytime and town averages are very noisy (only 6 routes each). 
You could improve the estimation by repeating the longest6 evaluation a couple of times, if you care about them for some reason, but usually we don't do anything with those values anyway.

If you are asking whether it is possible to make CARLA evaluations deterministic, then no, in the simulator version 0.9.10.1 CARLA has inherent randomness that you can not seed.
The newer versions are a bit better, but it still doesn't work perfectly.

closed 2025-03-17 by Kait0