autonomousvision/transfuser #287: Issue while setting up the environment

hamdaan-shaikh, 2025-03-12:
Not sure if others are facing the same issue while trying to set up the environment

I was following the steps to set up the repo

this line did not work for me: pip install torch-scatter -f https://data.pyg.org/whl/torch-1.11.0+cu102.html

This was the error:

(tfuse) developer@HULK:/data_nvme/end-to-end-autonomous-driving/transfuser$ pip install torch-scatter -f https://data.pyg.org/whl/torch-1.11.0+cu102.html
Looking in links: https://data.pyg.org/whl/torch-1.11.0+cu102.html
Collecting torch-scatter
  Using cached torch_scatter-2.1.1.tar.gz (107 kB)
Building wheels for collected packages: torch-scatter
  Building wheel for torch-scatter (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /home/developer/anaconda3/envs/tfuse/bin/python -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-s37hdp3k/torch-scatter_ee86740270ae4a27ba6e92327e685a51/setup.py'"'"'; __file__='"'"'/tmp/pip-install-s37hdp3k/torch-scatter_ee86740270ae4a27ba6e92327e685a51/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-um3iojw4
       cwd: /tmp/pip-install-s37hdp3k/torch-scatter_ee86740270ae4a27ba6e92327e685a51/
  Complete output (73 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-x86_64-3.7
  creating build/lib.linux-x86_64-3.7/torch_scatter
  copying torch_scatter/testing.py -> build/lib.linux-x86_64-3.7/torch_scatter
  copying torch_scatter/placeholder.py -> build/lib.linux-x86_64-3.7/torch_scatter
  copying torch_scatter/segment_csr.py -> build/lib.linux-x86_64-3.7/torch_scatter
  copying torch_scatter/utils.py -> build/lib.linux-x86_64-3.7/torch_scatter
  copying torch_scatter/scatter.py -> build/lib.linux-x86_64-3.7/torch_scatter
  copying torch_scatter/__init__.py -> build/lib.linux-x86_64-3.7/torch_scatter
  copying torch_scatter/segment_coo.py -> build/lib.linux-x86_64-3.7/torch_scatter
  creating build/lib.linux-x86_64-3.7/torch_scatter/composite
  copying torch_scatter/composite/std.py -> build/lib.linux-x86_64-3.7/torch_scatter/composite
  copying torch_scatter/composite/softmax.py -> build/lib.linux-x86_64-3.7/torch_scatter/composite
  copying torch_scatter/composite/logsumexp.py -> build/lib.linux-x86_64-3.7/torch_scatter/composite
  copying torch_scatter/composite/__init__.py -> build/lib.linux-x86_64-3.7/torch_scatter/composite
  running egg_info
  writing torch_scatter.egg-info/PKG-INFO
  writing dependency_links to torch_scatter.egg-info/dependency_links.txt
  writing requirements to torch_scatter.egg-info/requires.txt
  writing top-level names to torch_scatter.egg-info/top_level.txt
  reading manifest file 'torch_scatter.egg-info/SOURCES.txt'
  reading manifest template 'MANIFEST.in'
  warning: no previously-included files matching '*' found under directory 'test'
  adding license file 'LICENSE'
  writing manifest file 'torch_scatter.egg-info/SOURCES.txt'
  running build_ext
  Traceback (most recent call last):
    File "<string>", line 1, in <module>
    File "/tmp/pip-install-s37hdp3k/torch-scatter_ee86740270ae4a27ba6e92327e685a51/setup.py", line 140, in <module>
      include_package_data=include_package_data,
    File "/home/developer/anaconda3/envs/tfuse/lib/python3.7/site-packages/setuptools/__init__.py", line 87, in setup
      return distutils.core.setup(**attrs)
    File "/home/developer/anaconda3/envs/tfuse/lib/python3.7/site-packages/setuptools/_distutils/core.py", line 148, in setup
      return run_commands(dist)
    File "/home/developer/anaconda3/envs/tfuse/lib/python3.7/site-packages/setuptools/_distutils/core.py", line 163, in run_commands
      dist.run_commands()
    File "/home/developer/anaconda3/envs/tfuse/lib/python3.7/site-packages/setuptools/_distutils/dist.py", line 967, in run_commands
      self.run_command(cmd)
    File "/home/developer/anaconda3/envs/tfuse/lib/python3.7/site-packages/setuptools/dist.py", line 1214, in run_command
      super().run_command(command)
    File "/home/developer/anaconda3/envs/tfuse/lib/python3.7/site-packages/setuptools/_distutils/dist.py", line 986, in run_command
      cmd_obj.run()
    File "/home/developer/anaconda3/envs/tfuse/lib/python3.7/site-packages/wheel/bdist_wheel.py", line 299, in run
      self.run_command('build')
    File "/home/developer/anaconda3/envs/tfuse/lib/python3.7/site-packages/setuptools/_distutils/cmd.py", line 313, in run_command
      self.distribution.run_command(command)
    File "/home/developer/anaconda3/envs/tfuse/lib/python3.7/site-packages/setuptools/dist.py", line 1214, in run_command
      super().run_command(command)
    File "/home/developer/anaconda3/envs/tfuse/lib/python3.7/site-packages/setuptools/_distutils/dist.py", line 986, in run_command
[... 85 lines cut ...]
        super().run_command(command)
      File "/home/developer/anaconda3/envs/tfuse/lib/python3.7/site-packages/setuptools/_distutils/dist.py", line 986, in run_command
        cmd_obj.run()
      File "/home/developer/anaconda3/envs/tfuse/lib/python3.7/site-packages/setuptools/_distutils/command/build.py", line 135, in run
        self.run_command(cmd_name)
      File "/home/developer/anaconda3/envs/tfuse/lib/python3.7/site-packages/setuptools/_distutils/cmd.py", line 313, in run_command
        self.distribution.run_command(command)
      File "/home/developer/anaconda3/envs/tfuse/lib/python3.7/site-packages/setuptools/dist.py", line 1214, in run_command
        super().run_command(command)
      File "/home/developer/anaconda3/envs/tfuse/lib/python3.7/site-packages/setuptools/_distutils/dist.py", line 986, in run_command
        cmd_obj.run()
      File "/home/developer/anaconda3/envs/tfuse/lib/python3.7/site-packages/setuptools/command/build_ext.py", line 79, in run
        _build_ext.run(self)
      File "/home/developer/anaconda3/envs/tfuse/lib/python3.7/site-packages/setuptools/_distutils/command/build_ext.py", line 339, in run
        self.build_extensions()
      File "/home/developer/anaconda3/envs/tfuse/lib/python3.7/site-packages/torch/utils/cpp_extension.py", line 410, in build_extensions
        self._check_cuda_version()
      File "/home/developer/anaconda3/envs/tfuse/lib/python3.7/site-packages/torch/utils/cpp_extension.py", line 787, in _check_cuda_version
        raise RuntimeError(CUDA_MISMATCH_MESSAGE.format(cuda_str_version, torch.version.cuda))
    RuntimeError:
    The detected CUDA version (11.1) mismatches the version that was used to compile
    PyTorch (10.2). Please make sure to use the same CUDA versions.
    
    ----------------------------------------
ERROR: Command errored out with exit status 1: /home/developer/anaconda3/envs/tfuse/bin/python -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-s37hdp3k/torch-scatter_ee86740270ae4a27ba6e92327e685a51/setup.py'"'"'; __file__='"'"'/tmp/pip-install-s37hdp3k/torch-scatter_ee86740270ae4a27ba6e92327e685a51/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-fm6kn8he/install-record.txt --single-version-externally-managed --compile --install-headers /home/developer/anaconda3/envs/tfuse/include/python3.7m/torch-scatter Check the logs for full command output.

Kait0, 2025-03-12:
Just comment [this line](https://github.com/autonomousvision/transfuser/blob/3852e65e070f2efcfaba18fbbdb48057096b8a9e/team_code_transfuser/model.py#L11) and ignore torch-scatter if you have trouble setting it up. 99% you won't need it.

hamdaan-shaikh, 2025-03-12:
So it seems to be a CUDA mismatch while setting up the repo afresh. The detected CUDA version (11.1) mismatches the version that was used to compile PyTorch (10.2). Please make sure to use the same CUDA versions.

To work around this, I went to https://data.pyg.org/whl/torch-1.11.0+cu102.html and manually installed torch_scatter-2.0.9-cp37-cp37m-linux_x86_64.whl, is it a right way?

hamdaan-shaikh, 2025-03-12:
Thanks! I really appreciate it!

closed 2025-03-14 by Kait0