autonomousvision/transfuser #289: Issue while running the Carla server on WSL

amanysh99, 2025-03-14:
The error that I am encountering (Segmentation fault (core dumped)) indicates that the Carla simulator is crashing when trying to run in WSL. This is a common issue because WSL does not natively support graphical applications like the Unreal Engine (which Carla is built on).
So how can I run Carla server on WSL: ./CarlaUE4.sh --world-port=2000 -opengl

[image]

amanysh99, 2025-03-14:
And while trying to directly open the Carla.exe as Win version it stopped carla and give me timeout error

[image]

Kait0, 2025-03-14:
I don't see the point of using Windows subsystem for Linux.
CARLA has [Windows builds](https://github.com/carla-simulator/carla/releases/tag/0.9.10.1) and the code is python so runs on Windows as well.
You could also run the code within WSL and lets CARLA run in Windows which should work since they communicate via TCP.

amanysh99, 2025-03-14:
This is okey to run over the CMD?
./leaderboard/scripts/local_evaluation.sh <carla root> <working directory of this repo (*/transfuser/)>
I thought that due to .sh extensions I should use WSL or ubunto in general

Kait0, 2025-03-14:
Just look into the local_evaluation.sh script and write a windows version of it, it is a simple script that sets some environments variables and starts a python process.

closed 2025-03-15 by amanysh99