autonomousvision/transfuser #290: Visualization issue

amanysh99, 2025-03-16:
Despite setting the debug flag to True and ensuring that the specified save paths exist, no visualizations are being saved to the following directories:

D:\Lab6\transfuser\team_code_transfuser\save_folder\Visualizations\2\bev_lidar_{}

D:\Lab6\transfuser\team_code_transfuser\save_folder\Visualizations\2\image_with_lidar_{}

amanysh99, 2025-03-16:
Shall I run the Carla simulator while training?

Kait0, 2025-03-17:
no you don't need to run the CARLA sever.
Check with a debugger or print statement if the code enters [here](https://github.com/autonomousvision/transfuser/blob/3852e65e070f2efcfaba18fbbdb48057096b8a9e/team_code_transfuser/model.py#L796) and if not which variable is wrong.

closed 2025-03-20 by Kait0