autonomousvision/transfuser #291: Evaluation Issue

amanysh99, 2025-03-16:
After manually set the variables in CMD like this:
set CARLA_ROOT=D:\Lab6\CARLA_0.9.10\WindowsNoEditor
set WORK_DIR=D:\Lab6\transfuser
set PYTHONPATH=%PYTHONPATH%;%CARLA_ROOT%\PythonAPI
set PYTHONPATH=%PYTHONPATH%;%CARLA_ROOT%\PythonAPI\carla
set PYTHONPATH=%PYTHONPATH%;%CARLA_ROOT%\PythonAPI\carla\dist\carla-0.9.10-py3.7-win-amd64.egg
set SCENARIO_RUNNER_ROOT=%WORK_DIR%\scenario_runner
set LEADERBOARD_ROOT=%WORK_DIR%\leaderboard
set PYTHONPATH=%CARLA_ROOT%\PythonAPI\carla\;%SCENARIO_RUNNER_ROOT%;%LEADERBOARD_ROOT%;%PYTHONPATH%

set SCENARIOS=%WORK_DIR%\leaderboard\data\longest6\eval_scenarios.json
set ROUTES=%WORK_DIR%\leaderboard\data\longest6\longest6.xml
set REPETITIONS=1
set CHALLENGE_TRACK_CODENAME=SENSORS
set CHECKPOINT_ENDPOINT=%WORK_DIR%\results\transfuser_longest6.json
set TEAM_AGENT=%WORK_DIR%\team_code_transfuser\submission_agent.py
set TEAM_CONFIG=D:\Lab6\transfuser\model_ckpt\transfuser
set DEBUG_CHALLENGE=0
set RESUME=1
set DATAGEN=0

And running the evaluation script like this in CMD:
python %LEADERBOARD_ROOT%\leaderboard\leaderboard_evaluator_local.py ^
--scenarios=%SCENARIOS% ^
--routes=%ROUTES% ^
--repetitions=%REPETITIONS% ^
--track=%CHALLENGE_TRACK_CODENAME% ^
--checkpoint=%CHECKPOINT_ENDPOINT% ^
--agent=%TEAM_AGENT% ^
--agent-config=%TEAM_CONFIG% ^
--debug=%DEBUG_CHALLENGE% ^
--resume=%RESUME%

The args.txt:
{
  "id": "transfuser",
  "epochs": 41,
  "lr": 0.0001,
  "batch_size": 1,
  "logdir": "D:\\Lab6\\transfuser\\logs\\transfuser",
  "load_file": null,
  "start_epoch": 0,
  "setting": "all",
  "root_dir": "data",
  "schedule": 1,
  "backbone": "transFuser",
  "schedule_reduce_epoch_01": 30,
  "schedule_reduce_epoch_02": 40,
  "image_architecture": "regnety_032",
  "lidar_architecture": "regnety_032",
  "use_velocity": 0,
  "n_layer": 4,
  "wp_only": 0,
  "use_target_point_image": 1,
  "use_point_pillars": 0,
  "parallel_training": 0,
  "val_every": 5,
  "no_bev_loss": 0,
  "sync_batch_norm": 0,
  "zero_redundancy_optimizer": 0,
  "use_disk_cache": 0
}
And Carla simulator is already running.
It gives me the output in attachment after simple change in the Carla simulator like the weather is raining.

[image]

amanysh99, 2025-03-17:
[image]

Kait0, 2025-03-17:
delete the %WORK_DIR%\results\transfuser_longest6.json file and try again, the code probably thinks the route was already evaluated.
or use  set RESUME=0

amanysh99, 2025-03-17:
Thanks for your quick response, while doing that it returns:

[image]

Kait0, 2025-03-17:
strange. did you modify the code? the ego model should be [here](https://github.com/autonomousvision/transfuser/blob/3852e65e070f2efcfaba18fbbdb48057096b8a9e/team_code_transfuser/submission_agent.py#L663)

amanysh99, 2025-03-17:
Yes, I had mistakenly commented out that block of code earlier, but now it is running. However, it is not functioning properly and is producing the following output:
========= Preparing RouteScenario_0 (repetition 0) =========
> Setting up the agent
D:\Lab6\transfuser\model_ckpt\transfuser\best_model.pth
D:\Lab6\transfuser\model_ckpt\transfuser\model_1.pth
D:\Lab6\transfuser\model_ckpt\transfuser\model_10.pth
D:\Lab6\transfuser\model_ckpt\transfuser\model_11.pth
D:\Lab6\transfuser\model_ckpt\transfuser\model_12.pth
D:\Lab6\transfuser\model_ckpt\transfuser\model_13.pth
D:\Lab6\transfuser\model_ckpt\transfuser\model_14.pth
D:\Lab6\transfuser\model_ckpt\transfuser\model_15.pth
D:\Lab6\transfuser\model_ckpt\transfuser\model_16.pth
D:\Lab6\transfuser\model_ckpt\transfuser\model_17.pth
D:\Lab6\transfuser\model_ckpt\transfuser\model_18.pth
D:\Lab6\transfuser\model_ckpt\transfuser\model_19.pth
D:\Lab6\transfuser\model_ckpt\transfuser\model_2.pth
D:\Lab6\transfuser\model_ckpt\transfuser\model_20.pth
D:\Lab6\transfuser\model_ckpt\transfuser\model_21.pth
Watchdog exception - Timeout of 61.0 seconds occurred

Could not set up the required agent:
> Timeout: Agent took too long to setup

Traceback (most recent call last):
  File "D:\Lab6\transfuser\leaderboard\leaderboard\leaderboard_evaluator_local.py", line 272, in _load_and_run_scenario
    self.agent_instance = getattr(self.module_agent, agent_class_name)(args.agent_config)
  File "D:\Lab6\transfuser\leaderboard\leaderboard\autoagents\autonomous_agent.py", line 45, in __init__
    self.setup(path_to_conf_file)
  File "D:\Lab6\transfuser\team_code_transfuser\submission_agent.py", line 91, in setup
    net = LidarCenterNet(self.config, 'cpu', self.backbone, image_architecture, lidar_architecture, use_velocity)
  File "D:\Lab6\transfuser\team_code_transfuser\model.py", line 566, in __init__
    self._model = TransfuserBackbone(config, image_architecture, lidar_architecture, use_velocity=use_velocity).to(self.device)
  File "D:\Lab6\transfuser\team_code_transfuser\transfuser.py", line 90, in __init__
    config=config, use_velocity=use_velocity)
  File "D:\Lab6\transfuser\team_code_transfuser\transfuser.py", line 322, in __init__
    self.apply(self._init_weights)
  File "C:\Users\user\anaconda3\envs\tfuse\lib\site-packages\torch\nn\modules\module.py", line 667, in apply
    module.apply(fn)
  File "C:\Users\user\anaconda3\envs\tfuse\lib\site-packages\torch\nn\modules\module.py", line 667, in apply
    module.apply(fn)
  File "C:\Users\user\anaconda3\envs\tfuse\lib\site-packages\torch\nn\modules\module.py", line 667, in apply
    module.apply(fn)
  [Previous line repeated 1 more time]
  File "C:\Users\user\anaconda3\envs\tfuse\lib\site-packages\torch\nn\modules\module.py", line 668, in apply
    fn(self)
  File "D:\Lab6\transfuser\team_code_transfuser\transfuser.py", line 326, in _init_weights
    module.weight.data.normal_(mean=self.config.gpt_linear_layer_init_mean, std=self.config.gpt_linear_layer_init_std)
  File "D:\Lab6\transfuser\leaderboard\leaderboard\leaderboard_evaluator_local.py", line 114, in _signal_handler
    raise RuntimeError("Timeout: Agent took too long to setup")
RuntimeError: Timeout: Agent took too long to setup
> Registering the route statistics

========= Preparing RouteScenario_1 (repetition 0) =========
> Setting up the agent
D:\Lab6\transfuser\model_ckpt\transfuser\best_model.pth
D:\Lab6\transfuser\model_ckpt\transfuser\model_1.pth
D:\Lab6\transfuser\model_ckpt\transfuser\model_10.pth
D:\Lab6\transfuser\model_ckpt\transfuser\model_11.pth
D:\Lab6\transfuser\model_ckpt\transfuser\model_12.pth
D:\Lab6\transfuser\model_ckpt\transfuser\model_13.pth
D:\Lab6\transfuser\model_ckpt\transfuser\model_14.pth
D:\Lab6\transfuser\model_ckpt\transfuser\model_15.pth
D:\Lab6\transfuser\model_ckpt\transfuser\model_16.pth
D:\Lab6\transfuser\model_ckpt\transfuser\model_17.pth
D:\Lab6\transfuser\model_ckpt\transfuser\model_18.pth
D:\Lab6\transfuser\model_ckpt\transfuser\model_19.pth
D:\Lab6\transfuser\model_ckpt\transfuser\model_2.pth
D:\Lab6\transfuser\model_ckpt\transfuser\model_20.pth
D:\Lab6\transfuser\model_ckpt\transfuser\model_21.pth
D:\Lab6\transfuser\model_ckpt\transfuser\model_22.pth
Watchdog exception - Timeout of 61.0 seconds occurred

Could not set up the required agent:
> Timeout: Agent took too long to setup

amanysh99, 2025-03-17:
It is just represent the bev_view:

[image]

amanysh99, 2025-03-17:
I am really appreciate your hard work and your quickly response and sharing knowledge, Thank you!

Kait0, 2025-03-18:
so your code is running now? From the image it seems the model is not behaving correctly. I suggest you try the provided model weights first to check that the code is correct now.

amanysh99, 2025-03-18:
Yes, I understand that the code is not functioning properly. However, could you please clarify what you mean by "the provided weights"? Additionally, why isn’t the code displaying the LiDAR view?

amanysh99, 2025-03-18:
That while using the pre-trained model in that path: "D:\Lab6\transfuser\model_ckpt\transfuser\best_model.pth", with args.txt

[image]

Kait0, 2025-03-18:
We provide the model weights we used in the paper [here](https://github.com/autonomousvision/transfuser?tab=readme-ov-file#evaluation) you seemed to be using your own trained model.

What you show is the CARLA spectator camera. To get insight into the model you should use the visualization function.
Yes, you need to set SAVE_PATH environment variable and put debug to true. The visu function is called [here](https://github.com/autonomousvision/transfuser/blob/3852e65e070f2efcfaba18fbbdb48057096b8a9e/team_code_transfuser/model.py#L726C18-L726C27).

amanysh99, 2025-03-18:
I'am already download and extract a model file with 5 models (aim, cilrs, geometric_fusion, late_fusion, transfuser) in model_ckpt, or the model weights that you mean is the Zipped file called "model_2022"?

Kait0, 2025-03-20:
yes.
its an environment variable, you set it like the others in the top post.

amanysh99, 2025-03-20:
I set the SAVE_PATH like this: set SAVE_PATH=D:\Lab6\transfuser\team_code_transfuser\save_folder\Visualizations, and I also set the debug variable in the visualize function to true. However, it still only shows the BEV!
And there are no visualizations saved!

Kait0, 2025-03-20:
well check with a debug or print if the function linked above is called.

amanysh99, 2025-03-20:
It gives me that, but I set debug=True!

Debug: False, Save Path: D:\Lab6\transfuser\team_code_transfuser\save_folder\Visualizations, Step: 0
Debug: False, Save Path: D:\Lab6\transfuser\team_code_transfuser\save_folder\Visualizations, Step: 0
Debug: False, Save Path: D:\Lab6\transfuser\team_code_transfuser\save_folder\Visualizations, Step: 1
Debug: False, Save Path: D:\Lab6\transfuser\team_code_transfuser\save_folder\Visualizations, Step: 1
Debug: False, Save Path: D:\Lab6\transfuser\team_code_transfuser\save_folder\Visualizations, Step: 1
Debug: False, Save Path: D:\Lab6\transfuser\team_code_transfuser\save_folder\Visualizations, Step: 2
Debug: False, Save Path: D:\Lab6\transfuser\team_code_transfuser\save_folder\Visualizations, Step: 2
Debug: False, Save Path: D:\Lab6\transfuser\team_code_transfuser\save_folder\Visualizations, Step: 2
Debug: False, Save Path: D:\Lab6\transfuser\team_code_transfuser\save_folder\Visualizations, Step: 3
Debug: False, Save Path: D:\Lab6\transfuser\team_code_transfuser\save_folder\Visualizations, Step: 3
Debug: False, Save Path: D:\Lab6\transfuser\team_code_transfuser\save_folder\Visualizations, Step: 3
Debug: False, Save Path: D:\Lab6\transfuser\team_code_transfuser\save_folder\Visualizations, Step: 4
Debug: False, Save Path: D:\Lab6\transfuser\team_code_transfuser\save_folder\Visualizations, Step: 4
Debug: False, Save Path: D:\Lab6\transfuser\team_code_transfuser\save_folder\Visualizations, Step: 4
Debug: False, Save Path: D:\Lab6\transfuser\team_code_transfuser\save_folder\Visualizations, Step: 5
Debug: False, Save Path: D:\Lab6\transfuser\team_code_transfuser\save_folder\Visualizations, Step: 5
Debug: False, Save Path: D:\Lab6\transfuser\team_code_transfuser\save_folder\Visualizations, Step: 5
Debug: False, Save Path: D:\Lab6\transfuser\team_code_transfuser\save_folder\Visualizations, Step: 6
Debug: False, Save Path: D:\Lab6\transfuser\team_code_transfuser\save_folder\Visualizations, Step: 6
Debug: False, Save Path: D:\Lab6\transfuser\team_code_transfuser\save_folder\Visualizations, Step: 6

[image]

amanysh99, 2025-03-20:
I solved by temporarily hardcode debug=True in the forward_ego method 

[image]

And it saved the images with TopDown, BEV, front camera in the save_path, but not represent all that view here:

[image]

Kait0, 2025-03-20:
that is normal the CARLA spectator BEV camera is independent of the visualization code which saves to disk