autonomousvision/transfuser #292: Longest6 Benchmark VS Town05

MCUBE-2023, 2025-03-19:
Dear author, 

I am quite confused about the difference between the Longest6 Benchmark compared to Town05.

So referring the Longest6 Benchmark link: https://github.com/autonomousvision/transfuser/tree/2022/leaderboard/data/longest6
and referring to first attached screenshot, I can understand that Longest6 Benchmark is made of the combination between 6 routes and 6 weather conditions for each route, which makes 36 routes in total for the Longest6 Benchmark. However, the Town05 contains 32 routes. To this end, I have the following questions: 

[image]

1- So if I am going to evaluate Town05 performance using the Longest6 Benchmark, would the evaluation be restricted to the 32 routes of the Longest6 Benchmark as illustrated in the 2nd screenshot ? 

[image]

2- Referring to this link: https://github.com/autonomousvision/transfuser/blob/2022/leaderboard/data/town_maps_xodr/Town05.png
and based on the third attached screenshot, What does this image of Town05 represents exactly ? Is it one of the images of the 36 routes presented in the first screenshot (see the fourth and the fifth screenshost to understand my potential confusion between the image of town05 and the image of the 36 routes)  ?

[image]

[image]

Kait0, 2025-03-20:
First of all longest6 and Town05 long (I assume with Town 05 you mean the Town 05 long benchmark) are different benchmarks, they don't necessarily have anything to do with each other.

longest6 is 36 unique routes as you can see in your image not 6. each route has a different weather. Its 6 routes per town (Town 1-6).
Town05 long only uses routes in Town 05 as the name suggests and only has a single weather. Town05 long has [10 different routes](https://github.com/autonomousvision/transfuser/blob/cvpr2021/leaderboard/data/evaluation_routes/routes_town05_long.xml) I don't know why you think its 32.

Also Town05 long uses its own procedurally [generated scenario definitions](https://github.com/autonomousvision/transfuser/blob/cvpr2021/leaderboard/data/scenarios/town05_all_scenarios.json).
Whereas longest6 uses [handcrafted scenarios](https://github.com/autonomousvision/transfuser/blob/2022/leaderboard/data/longest6/eval_scenarios.json) from the CARLA team.

Additionally longest6 needs increased traffic density and doesn't consider stop signs (check [the documentation](https://github.com/autonomousvision/transfuser/tree/2022/leaderboard)). Town 05 does not apply these changes.

1. The image you show has nothing to do with Town 05 long. Also these are different towns as you can see from the shape of the map.

2. This is the Map of Town05 without any routes rendered in. The 5th row in the grey image renders 6 routes on top of the Town05 map.

Or is your goal to only evaluate on the 6x Town 05 routes of longest6?

MCUBE-2023, 2025-03-20:
Thank you for your fast response.

For my evaluation, I am, using routes_town05_short.xml which contains 32 routes (see the link below). To this end, I have these questions: 

1- Can I use "routes_town05_short.xml" as benchmark ?
2-  Similarly to the figures shown for each routes of Long6, can I see a figure of each route of the 32 routes of routes_town05_short.xml (for example a figure that shows the starting point of the ego-vehicle in green and the end point in red) as illustrated in this image below ?

[image]

https://github.com/autonomousvision/transfuser/blob/cvpr2021/leaderboard/data/validation_routes/routes_town05_short.xml

Kait0, 2025-03-21:
1. Yes, its called Town05 Short in the CVPR paper. Its an easier benchmark and mostly considered solved nowadays I think. You also need to use the same scenarios for it as for Town05 long.
2. Have a look at this script https://github.com/autonomousvision/transfuser/tree/2022/tools/dataset#visualization-tool

MCUBE-2023, 2025-03-21:
Thank you for your quick reply. 
I will be grateful for answering this issue as well please: https://github.com/autonomousvision/transfuser/issues/293#issue-2936719607

MCUBE-2023, 2025-03-22:
I also have these question for the attached image:
1- Do the blue points represent the "ground truth" waypoints for the ego-vehicle for the corresponding route ?
2- What do the red point and red green point represent ? 

[image]

Kait0, 2025-03-23:
The points are intermediate goals positions that the ego vehicle is supposed to pass through.
Green and Red is just color coded for start and end I think.

MCUBE-2023, 2025-03-27:
Thank you so much for your quick answers ! I am grateful for that !

closed 2025-03-27 by MCUBE-2023