autonomousvision/transfuser #293: Cross-Modal Attention Statistics Reporting

MCUBE-2023, 2025-03-20:
Dear authors,
I have a follow-up question regarding Table 4 (Cross-Modal Attention Statistics) in your article. Are you reporting these statistics for one route (for example Route_0) ? Or are they reported as an average for a couple of routes (for example, for Twon05 short, you are computing these statistics as an average of the 32 routes) ?

[image]

ap229997, 2025-03-23:
The statistics are computed across 1000 randomly sampled examples from multiple routes. I do not remember which all routes and towns were used for this table.

MCUBE-2023, 2025-03-27:
Thank you so much for your answer !

closed 2025-03-27 by MCUBE-2023