autonomousvision/transfuser #298: FileNotFoundError: Missing lidar directory in dataset root while running Train.py

Raz-o, 2025-04-02:
Hello,  

I am trying to train the TransFuser model using my own dataset as well as the provided dataset, but I keep encountering a `FileNotFoundError` related to the `lidar` directory. The error occurs regardless of the dataset I use.  

Steps to Reproduce:  
1. **Run the following command:**  

   ```sh
   (tfuse) ralj@ralj:~/transfuser/team_code_transfuser$ python train.py --batch_size 10 \
       --logdir /home/ralj/transfuser/model \
       --root_dir /home/ralj/transfuser/data/mytown \
       --parallel_training 0
   ```  

2. **Training starts but fails due to a missing `lidar` directory.**  

Error Output:  

```
Start method of multiprocessing: fork
Total trainable parameters:  168018327
  0%|                                                                                                                                                                                | 0/23 [00:00<?, ?it/s]
{
  "message": {
    "message": "FileNotFoundError: [Errno 2] No such file or directory: '/home/ralj/transfuser/data/mytown/Routes_Scenario1_Town03_curved_Seed0/Scenario1_Town03_curved_route22_11_23_22_20_11/semantics/lidar'",
    "extraInfo": {
      "py_callstack": "Traceback (most recent call last):\n  File \"/home/ralj/miniconda3/envs/tfuse/lib/python3.7/site-packages/torch/distributed/elastic/multiprocessing/errors/__init__.py\", line 345, in wrapper\n    return f(*args, **kwargs)\n  File \"train.py\", line 150, in main\n    train_set = CARLA_Data(root=config.train_data, config=config, shared_dict=shared_dict)\n  File \"/home/ralj/transfuser/team_code_transfuser/data.py\", line 54, in __init__\n    num_seq = len(os.listdir(route_dir / \"lidar\"))\nFileNotFoundError: [Errno 2] No such file or directory: '/home/ralj/transfuser/data/mytown/Routes_Scenario1_Town03_curved_Seed0/Scenario1_Town03_curved_route22_11_23_22_20_11/semantics/lidar'\n",
      "timestamp": "1743599827"
    }
  }
}
Traceback (most recent call last):
  File "train.py", line 399, in <module>
    main()
  File "/home/ralj/miniconda3/envs/tfuse/lib/python3.7/site-packages/torch/distributed/elastic/multiprocessing/errors/__init__.py", line 345, in wrapper
    return f(*args, **kwargs)
  File "train.py", line 150, in main
    train_set = CARLA_Data(root=config.train_data, config=config, shared_dict=shared_dict)
  File "/home/ralj/transfuser/team_code_transfuser/data.py", line 54, in __init__
    num_seq = len(os.listdir(route_dir / "lidar"))
FileNotFoundError: [Errno 2] No such file or directory: '/home/ralj/transfuser/data/mytown/Routes_Scenario1_Town03_curved_Seed0/Scenario1_Town03_curved_route22_11_23_22_20_11/semantics/lidar'
(tfuse) ralj@ralj:~/transfuser/team_code_transfuser$
```

**Additional Information:**  

- The dataset was generated using CARLA, and I ensured that the files were copied correctly.  
- I also downloaded the provided dataset, but the error persists.  
- The `semantics/lidar` directory seems to be missing from the dataset structure.  
- I couldn't find any related issues on GitHub or documentation explaining the expected folder structure.  

Could you clarify the correct dataset organization, or let me know if this is a bug? If possible, providing a brief text file with the required directory structure would be very helpful.  

Thanks in advance!

Kait0, 2025-04-02:
> /home/ralj/transfuser/data/mytown/Routes_Scenario1_Town03_curved_Seed0/Scenario1_Town03_curved_route22_11_23_22_20_11/semantics/lidar

This should be /home/ralj/transfuser/data/mytown/Routes_Scenario1_Town03_curved_Seed0/Scenario1_Town03_curved_route22_11_23_22_20_11/lidar 

instead I believe (without the semantics). You probably set the wrong folder as root, try: 
--root_dir /home/ralj/transfuser/data

You are using the 2022 branch of the repo? (Otherwise maybe [this](https://github.com/autonomousvision/transfuser/issues/92) is helpful).

Raz-o, 2025-04-02:
Thanks for the suggestion! I’ve already tried using --root_dir /home/ralj/transfuser/data and it is creating the model, and I’m fairly certain I’m using the 2022 repo as I only got into the code about a month ago.

Regarding the data generation for just Town 3, Scenario 1, I have already generated the data for that specific scenario. Could you clarify how I can train a model using just that dataset for experimentation? Should I just point the --root_dir to the directory where the Town 3, Scenario 1 data is stored or i have to do multiple scenarios together?

Thanks in advance!

Kait0, 2025-04-02:
Yes, you use the root dir option to use a new dataset.

The code is searching for the lidar folder 1 floor too deep in your folder structure.
Most likely, one folder level is missing.
You can debug in the [dataloader](https://github.com/autonomousvision/transfuser/blob/3852e65e070f2efcfaba18fbbdb48057096b8a9e/team_code_transfuser/data.py#L20) and [config](https://github.com/autonomousvision/transfuser/blob/3852e65e070f2efcfaba18fbbdb48057096b8a9e/team_code_transfuser/config.py#L206) why that is the case.

Raz-o, 2025-04-03:
Thanks for the clarification!

I tried training with a batch size of 10, but it crashed, so I reduced it to 5. It took around 12 hours to train model_1 and its optimized file.

I have a few questions now:

**How many models is it actually building**? I see multiple files, and it's taking a lot of time. Do I need all of them, or can I just use a single one or train faster?

**After running local_evaluation.sh, do I get my new agent as a .py file?** If not, how do I extract and use my trained model for inference and its file to make a video like [this](https://github.com/autonomousvision/neat/blob/main/neat/assets/neat_clip.GIF)?

**How can I run my trained agent only in Town 3 and compare it against a basic model?** I want to evaluate its performance in isolation.

These are my main concerns for now. Any guidance would be really helpful. Thanks!

Kait0, 2025-04-04:
I recommend you look a bit through the closed issues about training with small batch size and single gpu.

You might want to increase [workers](https://github.com/autonomousvision/transfuser/blob/3852e65e070f2efcfaba18fbbdb48057096b8a9e/team_code_transfuser/train.py#L162) and use a different learning rate.

You want to train at least 31 models and use the last checkpoint.
You can check the documentation and issues for evaluation and visualization and look at the code for the [setting](https://github.com/autonomousvision/transfuser/blob/3852e65e070f2efcfaba18fbbdb48057096b8a9e/team_code_transfuser/train.py#L38) feature on how to withhold towns during training.

Raz-o, 2025-04-06:
I’ve already tried to resolve the issue based on previous discussions regarding training with a small batch size and using a single GPU, but I’m still facing some challenges.

Here’s what I’ve tried:

1. **Set `RANK` Manually**:  
   I attempted to manually set the `RANK` environment variable to `0` to resolve the `KeyError: 'RANK'` when running the training script. However, I still encounter the following error:

   ```
   KeyError: 'RANK'
   ```

   This error arises because the code expects the `RANK` variable to be set for distributed training, but I’m training on a single GPU, so distributed training shouldn't be necessary in my case.

2. **Disabling Distributed Training**:  
   I also tried commenting out line 95 in `submission.py` as suggested for single-GPU training, but I’m still facing the same issue.

3. **Training with `--setting Town03`**:  
   I attempted to use the `--setting Town03` to withhold Town03 during training, but this also didn’t work as expected.

Here’s the command I’m running:

```bash
python train.py \
--id transfuser_town03 \
--root_dir /home/ralj/transfuser/data \
--logdir /home/ralj/transfuser/model \
--batch_size 8 \
--lr 0.000033 \
--load_file /home/ralj/transfuser/model/pretransfuser
```

And here’s the result from running the command:

```
(tfuse) ralj@ralj:~/transfuser/team_code_transfuser$ python train.py \
--root_dir /home/ralj/transfuser/data \
--logdir /home/ralj/transfuser/model \
--batch_size 8 \
--lr 0.000033 \
--load_file /home/ralj/transfuser/model/pretransfuser
Start method of multiprocessing: fork
{
  "message": {
    "message": "KeyError: 'RANK'",
    "extraInfo": {
      "py_callstack": "Traceback (most recent call last):\n  File \"/home/ralj/miniconda3/envs/tfuse/lib/python3.7/site-packages/torch/distributed/elastic/multiprocessing/errors/__init__.py\", line 345, in wrapper\n    return f(*args, **kwargs)\n  File \"train.py\", line 95, in main\n    rank       = int(os.environ[\"RANK\"]) #Rank accross all processes\n  File \"/home/ralj/miniconda3/envs/tfuse/lib/python3.7/os.py\", line 681, in __getitem__\n    raise KeyError(key) from None\nKeyError: 'RANK'\n",
      "timestamp": "1743982191"
    }
  }
}
Traceback (most recent call last):
  File "train.py", line 399, in <module>
    main()
  File "/home/ralj/miniconda3/envs/tfuse/lib/python3.7/site-packages/torch/distributed/elastic/multiprocessing/errors/__init__.py", line 345, in wrapper
    return f(*args, **kwargs)
  File "train.py", line 95, in main
    rank       = int(os.environ["RANK"]) #Rank accross all processes
  File "/home/ralj/miniconda3/envs/tfuse/lib/python3.7/os.py", line 681, in __getitem__
    raise KeyError(key) from None
KeyError: 'RANK'
(tfuse) ralj@ralj:~/transfuser/team_code_transfuser$
```

I would appreciate any additional suggestions or clarifications to move forward with this. Specifically:
- Is there any way to fully disable distributed training for single-GPU use?
- Is there a correct way to use `--setting Town03` for withholding towns during training?
- Do i have to write out each name of the pretrained agent in the --load_file

Thanks in advance for your help!

Kait0, 2025-04-10:
--setting Town03

This feature is not implemented; you need to do that yourself.
Just read the code on how the other settings are implemented and adapt from that.

It is possible to start the multi-GPU training with 1 GPU by setting nthe umber of GPUs to 1. You don't need to set RANK manually. Multi-gpu training is started differently, as you can see in the README.

--load_file is for restarting crashed jobs, you use the latest agent for continuing training.

Raz-o, 2025-04-10:
Thanks a lot with the whole explanation,i understand it now, but lastly,do you think i can train for bad weathers using pretrained agents as the initial check point and training them further for better result??

Kait0, 2025-04-15:
in principle yes. Finetuning might be tricky with the provided checkpoints since their batch-norm weights are not trained.
You can check the closed issues.
I am not sure if you gain much by fine-tuning with bad weather. 
I did not have the impression that transfuser has problems with bad weather, except maybe nighttime.

closed 2025-04-23 by Kait0