autonomousvision/transfuser #302: Could you please tell me why it takes me so long to complete a scene?

ChuXX325, 2025-04-14:
Could you please tell me why it takes me so long to complete a scene?

[image]

[image]

Kait0, 2025-04-15:
36 minutes per route does not sound unusual.
You are inferencing a medium-sized model on top of an expensive simulation, and you are simulating 1-2 km of driving (at low speeds).

Kait0, 2025-04-15:
Model inference time should reduce since the 4090 computes faster.
CARLA time will likely be similar since it is more CPU-bound.
Maybe 50% faster overall, hard to predict.

Kait0, 2025-04-15:
see [here](https://github.com/autonomousvision/carla_garage/tree/leaderboard_1?tab=readme-ov-file#how-to-actually-evaluate) on how we evaluate.

ChuXX325, 2025-04-15:
I understand, thank you very much for your answer！

closed 2025-04-15 by ChuXX325