autonomousvision/transfuser #304: Unable to Submit Model to Leaderboard Alpha – Institution Access Required?

Raz-o, 2025-04-25:
Hi,
I’ve built and tested my Docker image locally using the leaderboard evaluation tools, and it performs really well. But when I tried submitting to Leaderboard Alpha, I got a message saying I need to contact my institution.

I’m short on time and would really appreciate any help or clarification on how to get access and proceed with submission to get a better evaluation on a better server.

Thanks!

Kait0, 2025-04-26:
The CARLA leaderboard switched from Alphadrive to Eval.AI.
But anyways it is currently closed you can't evaluate.
You can find the details on their website. https://leaderboard.carla.org/

Raz-o, 2025-04-28:
Thanks a lot for your help and support throughout the whole process,i will close this now.

closed 2025-04-28 by Raz-o