autonomousvision/transfuser #307: Evaluation Checkpoint issue

amanysh99, 2025-05-09:
Unfortunately, I am encountering a recurring issue while attempting to evaluate the model using the pre-trained weights in the CARLA leaderboard. The process consistently stalls at the "Registering the global statistics" stage, despite having resolved this problem once before. I am unsure of the root cause.

Step1:
set CARLA_ROOT=D:\Lab6\CARLA_0.9.10\WindowsNoEditor
set WORK_DIR=D:\Lab6\transfuser
set SAVE_PATH=D:\Lab6\transfuser\team_code_transfuser\save_folder\Visualizations
set PYTHONPATH=%PYTHONPATH%;%CARLA_ROOT%\PythonAPI
set PYTHONPATH=%PYTHONPATH%;%CARLA_ROOT%\PythonAPI\carla
set PYTHONPATH=%PYTHONPATH%;%CARLA_ROOT%\PythonAPI\carla\dist\carla-0.9.10-py3.7-win-amd64.egg
set SCENARIO_RUNNER_ROOT=%WORK_DIR%\scenario_runner
set LEADERBOARD_ROOT=%WORK_DIR%\leaderboard
set PYTHONPATH=%CARLA_ROOT%\PythonAPI\carla\;%SCENARIO_RUNNER_ROOT%;%LEADERBOARD_ROOT%;%PYTHONPATH%

set SCENARIOS=%WORK_DIR%\leaderboard\data\longest6\eval_scenarios.json
set ROUTES=%WORK_DIR%\leaderboard\data\longest6\longest6.xml
set REPETITIONS=1
set CHALLENGE_TRACK_CODENAME=SENSORS

set TEAM_AGENT=%WORK_DIR%\team_code_transfuser\submission_agent.py
set TEAM_CONFIG=%WORK_DIR%\model_ckpt\transfuser
set DEBUG_CHALLENGE=0
set RESUME=1
set DATAGEN=0

Step2:
python %LEADERBOARD_ROOT%\leaderboard\leaderboard_evaluator_local.py ^
--scenarios=%SCENARIOS% ^
--routes=%ROUTES% ^
--repetitions=%REPETITIONS% ^
--track=%CHALLENGE_TRACK_CODENAME% ^
--checkpoint=%CHECKPOINT_ENDPOINT% ^
--agent=%TEAM_AGENT% ^
--agent-config=%TEAM_CONFIG% ^
--debug=%DEBUG_CHALLENGE% ^
--resume=%RESUME%

[image]

Kait0, 2025-05-09:
set resume = 0 or delete your .json results file.
The software thinks you already finished the evaluation (perhaps due to a prior crashed run)

Kait0, 2025-05-09:
Delete the %CHECKPOINT_ENDPOINT% file or change  set RESUME=0

amanysh99, 2025-05-09:
[image]

Kait0, 2025-05-09:
that is the actual error, seems like your pytorch installation is missing cuda.
Note that the software stack in this repo is older and doesn't run with modern GPUs 3000 or newer out of the box.
You can update it yourself or use some of our [newer repos](https://github.com/autonomousvision/carla_garage) depending on what you want to do.

amanysh99, 2025-05-09:
For debugging purposes, I have updated all device assignments in submission_agent.py, changing them from 'cuda' to 'cpu' to ensure CPU-only execution but that didn't solve the problem and again i got . This configuration has previously run successfully without generating any CUDA errors.

[image]

amanysh99, 2025-05-09:
That attachment from last month where the evaluation was running successfully:

[image]

Kait0, 2025-05-09:
The error traceback you showed above still has a line with hardcoded 'cuda' in it 
In net=LidarCenterNet(..., 'cuda',...)
Perhaps you need to change that to cpu.

amanysh99, 2025-05-09:
I have changed all CUDA references to CPU, but I am still encountering the same error.

[image]

amanysh99, 2025-05-09:
I’m not sure if I understood you correctly, but the results folder is empty, and there are no .json files in my specified directory paths.
set CARLA_ROOT=D:\Lab6\CARLA_0.9.10\WindowsNoEditor
set WORK_DIR=D:\Lab6\transfuser
set SAVE_PATH=D:\Lab6\transfuser\team_code_transfuser\save_folder\Visualizations
set PYTHONPATH=%PYTHONPATH%;%CARLA_ROOT%\PythonAPI
set PYTHONPATH=%PYTHONPATH%;%CARLA_ROOT%\PythonAPI\carla
set PYTHONPATH=%PYTHONPATH%;%CARLA_ROOT%\PythonAPI\carla\dist\carla-0.9.10-py3.7-win-amd64.egg
set SCENARIO_RUNNER_ROOT=%WORK_DIR%\scenario_runner
set LEADERBOARD_ROOT=%WORK_DIR%\leaderboard
set PYTHONPATH=%CARLA_ROOT%\PythonAPI\carla\;%SCENARIO_RUNNER_ROOT%;%LEADERBOARD_ROOT%;%PYTHONPATH%

set SCENARIOS=%WORK_DIR%\leaderboard\data\longest6\eval_scenarios.json
set ROUTES=%WORK_DIR%\leaderboard\data\longest6\longest6.xml
set REPETITIONS=1
set CHALLENGE_TRACK_CODENAME=SENSORS

set TEAM_AGENT=%WORK_DIR%\team_code_transfuser\submission_agent.py
set TEAM_CONFIG=%WORK_DIR%\model_ckpt\transfuser
set DEBUG_CHALLENGE=0
set RESUME=0
set DATAGEN=0

python %LEADERBOARD_ROOT%\leaderboard\leaderboard_evaluator_local.py ^
--scenarios=%SCENARIOS% ^
--routes=%ROUTES% ^
--repetitions=%REPETITIONS% ^
--track=%CHALLENGE_TRACK_CODENAME% ^
--agent=%TEAM_AGENT% ^
--agent-config=%TEAM_CONFIG% ^
--debug=%DEBUG_CHALLENGE% ^
--resume=%RESUME%

Kait0, 2025-05-09:
You are not specifiying the --checkpoint option of leaderboard_evaluator_local.py

I mean the simulation_results.json file.
It will be stored somewhere. Probably in leaderboard/leaderboard/
[image]

amanysh99, 2025-05-09:
"I sincerely appreciate your assistance. It is now functioning properly again thanks to your support

closed 2025-05-11 by amanysh99