autonomousvision/transfuser #308: Channel Mismatch Error

amanysh99, 2025-05-21:
The specific error message I'm receiving is:
RuntimeError: Given groups=1, weight of size [32, 3, 3, 3], expected input[1, 2, 256, 256] to have 3 channels, but got 2 channels instead

This error occurs within the forward method of the TransfuserBackbone (in transfuser.py), specifically when the lidar_encoder._model.conv1 layer attempts to process the LiDAR input. The traceback points to D:\Lab6\transfuser\team_code_transfuser\transfuser.py", line 164.

From what I understand, the lidar_encoder is initialized to expect 3 input channels (likely due to self.config.use_target_point_image being True in the configuration, which adds one channel for the target point image to the standard 2 channels for LiDAR features). However, the lidar_tensor being passed to it currently only has 2 channels.

I've identified that the forward method's signature in transfuser.py is def forward(self, image, lidar, velocity):. While the CARLA environment provides target_point_image in its observation, it seems this is not being explicitly passed to or concatenated within this forward method.

My proposed solution is to:

Update the forward method's signature in transfuser.py to def forward(self, image, lidar, velocity, target_point_image=None):.
Concatenate target_point_image with lidar along the channel dimension before lidar_tensor is passed to the lidar_encoder.
I've made this change locally, and I believe it should address the channel mismatch by ensuring the lidar_encoder receives the expected 3 channels when use_target_point_image is enabled.

 File "D:\Lab6\transfuser\team_code_transfuser\transfuser.py", line 263, in get_policy_features
    _, _, fused_features_main = self.forward(rgb_obs, lidar_obs, ego_vel_obs)
  File "D:\Lab6\transfuser\team_code_transfuser\transfuser.py", line 168, in forward
    lidar_features = self.lidar_encoder._model.conv1(lidar_tensor)
  File "C:\Users\user\anaconda3\envs\tfuse\lib\site-packages\torch\nn\modules\module.py", line 1110, in _call_impl
    return forward_call(*input, **kwargs)
  File "C:\Users\user\anaconda3\envs\tfuse\lib\site-packages\torch\nn\modules\conv.py", line 447, in forward
    return self._conv_forward(input, self.weight, self.bias)
  File "C:\Users\user\anaconda3\envs\tfuse\lib\site-packages\torch\nn\modules\conv.py", line 444, in _conv_forward
    self.padding, self.dilation, self.groups)
RuntimeError: Given groups=1, weight of size [32, 3, 3, 3], expected input[1, 2, 256, 256] to have 3 channels, but got 2 channels instead

Kait0, 2025-05-22:
The target_point image is concatenated to the lidar [here](https://github.com/autonomousvision/transfuser/blob/3852e65e070f2efcfaba18fbbdb48057096b8a9e/team_code_transfuser/model.py#L742) and does not need to be passed to transfuser.py separately.

amanysh99, 2025-05-22:
I've confirmed that I am concatenating the target_point_image with the LiDAR input before passing it to transfuser.py, and I've ensured it's not being passed separately.

However, I am still encountering the same error: RuntimeError: Given groups=1, weight of size [32, 2, 3, 3], expected input[12, 3, 256, 256] to have 2 channels, but got 3 channels instead.     

Traceback (most recent call last):
  File "train_try101.py", line 1019, in <module>
    main()
  File "train_try101.py", line 177, in main
    trainer_il.train()
  File "train_try101.py", line 969, in train
    losses = self.load_data_compute_loss(data)
  File "train_try101.py", line 946, in load_data_compute_loss
    semantic=semantic, num_points=num_points)
  File "C:\Users\user\anaconda3\envs\tfuse\lib\site-packages\torch\nn\modules\module.py", line 1110, in _call_impl
    return forward_call(*input, **kwargs)
  File "D:\Lab6\transfuser\team_code_transfuser\model.py", line 765, in forward
    features, image_features_grid, fused_features = self._model(rgb, lidar_bev, ego_vel)
  File "C:\Users\user\anaconda3\envs\tfuse\lib\site-packages\torch\nn\modules\module.py", line 1110, in _call_impl
    return forward_call(*input, **kwargs)
  File "D:\Lab6\transfuser\team_code_transfuser\transfuser.py", line 163, in forward
    lidar_features = self.lidar_encoder._model.conv1(lidar_tensor)
  File "C:\Users\user\anaconda3\envs\tfuse\lib\site-packages\torch\nn\modules\module.py", line 1110, in _call_impl
    return forward_call(*input, **kwargs)
  File "C:\Users\user\anaconda3\envs\tfuse\lib\site-packages\torch\nn\modules\conv.py", line 447, in forward
    return self._conv_forward(input, self.weight, self.bias)
  File "C:\Users\user\anaconda3\envs\tfuse\lib\site-packages\torch\nn\modules\conv.py", line 444, in _conv_forward
    self.padding, self.dilation, self.groups)
RuntimeError: Given groups=1, weight of size [32, 2, 3, 3], expected input[12, 3, 256, 256] to have 2 channels, but got 3 channels instead

Kait0, 2025-05-26:
Well, you need to check why the code expects 2 channels even though you seem to want to use 3 (2 channels for [different heights](https://github.com/autonomousvision/transfuser/blob/3852e65e070f2efcfaba18fbbdb48057096b8a9e/team_code_transfuser/data.py#L467), 1 for target point).
You can check what the code does [here](https://github.com/autonomousvision/transfuser/blob/3852e65e070f2efcfaba18fbbdb48057096b8a9e/team_code_transfuser/transfuser.py#L26) and [here](https://github.com/autonomousvision/transfuser/blob/3852e65e070f2efcfaba18fbbdb48057096b8a9e/team_code_transfuser/transfuser.py#L477C39-L477C50).

amanysh99, 2025-05-26:
But what if I want to directly use a new LiDAR input from CARLA, like this: self.sensors['lidar'].listen(...)? Of course, this will cause a mismatch in the forward method, specifically in the line: lidar_features = self.lidar_encoder._model.conv1(lidar_tensor).

Kait0, 2025-05-26:
Well if you want to change something you need to change all relevant places in the code. Also you need to make sure the corrdinate systems are correct (they are quite messy in this codebase). 
You can have a look how the lidar is processed throughout the [submission_agent](https://github.com/autonomousvision/transfuser/blob/3852e65e070f2efcfaba18fbbdb48057096b8a9e/team_code_transfuser/submission_agent.py#L211) file.