autonomousvision/transfuser #309: Whether evaluation data can be accessed

Captain6606, 2025-05-27:
First of all, thank you for your outstanding work!

I am a beginner who has just entered this field.  I would like to know whether it is possible to obtain the input data (RGB and LiDAR) used for model evaluation when running local_evaluation.sh.  If it is possible, please let me know how to obtain this input data.  Thank you very much, and I look forward to your reply.

Captain6606, 2025-05-27:
If I only need to evaluate the model, do I still have to download this 210GB dataset?

Kait0, 2025-05-28:
Perhaps you can familiarize yourself a bit with the literature before diving in.
We offer a [survey](https://arxiv.org/abs/2306.16927) and [lecture](https://www.youtube.com/watch?v=_q4WUxgwDeg&list=PL05umP7R6ij321zzKXK6XCQXAaaYjQbzr) for that purpose.

There is no evaluation dataset, the model is evaluated in simulation.
If you only want to investigate evaluation, you do not need to download the training dataset.

Captain6606, 2025-05-28:
I'm really sorry, but I've already read your paper. I know there is no dataset for evaluation, but I'd like to know if I can obtain the sensor data in real-time during the simulation evaluation (this data would be very useful for my follow-up research).