autonomousvision/transfuser #311: Trying different backbone

amanysh99, 2025-06-13:
Why might this error be happening? I just tried a different backbone

[image]
[image]
[image]

Kait0, 2025-06-15:
Well as the first error indicates you tried a backbone that does not exist. You can go to the line where the expection is raise to see which ones are valid.

amanysh99, 2025-06-16:
[image]

Kait0, 2025-06-19:
You forgot to install the additional towns package of CARLA. See [here](https://github.com/autonomousvision/transfuser/blob/3852e65e070f2efcfaba18fbbdb48057096b8a9e/setup_carla.sh#L7).

Kait0, 2025-06-19:
You changed the benchmark to workaround a bug? You can't be serious...

How is CARLA supposed to find the maps if you just download them into a random folder? You need to install them.
You have access to google and chatgpt too.
This is documented e.g. [here](https://carla.readthedocs.io/en/0.9.10/start_quickstart/#import-additional-assets)
GitHub issues are not the place to ask these questions.

closed 2025-06-19 by Kait0