autonomousvision/transfuser #324: About running the agent

bigbigbabyer-ai, 2025-09-22:
Hello, thank you for your contributions. I am a beginner, and I encountered some issues while trying to reproduce your code. Regarding the “running the agent” part, after executing

./CarlaUE4.sh --world-port=2000 -opengl
and
./leaderboard/scripts/local_evaluation.sh ~/transfuser/carla ~/transfuser

the output is:

(tfuse) l@l:~/transfuser$ ./leaderboard/scripts/local_evaluation.sh ~/transfuser/carla ~/transfuser
Using TEAM_CONFIG=/home/l/transfuser/model_ckpt/models_2022/transfuser
/home/l/transfuser/leaderboard/leaderboard/leaderboard_evaluator_local.py:22: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
/home/l/transfuser/leaderboard/leaderboard/leaderboard_evaluator_local.py:89: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
  if LooseVersion(dist.version) < LooseVersion('0.9.10'):
> Registering the global statistics

The CARLA server only changes the environment, but no vehicles appear. Why is this happening, and how can I make the vehicles drive autonomously? I would appreciate your explanation.

Kait0, 2025-09-22:
delete the result.json file the code thinks all routes are finished already.
You can look through the issues this came a couple of times already.

akshay-ram-pp, 2025-09-22:
> delete the result.json file the code thinks all routes are finished already. You can look through the issues this came a couple of times already.

i was also facing a similar issue.
Can u please specify the path of result.json. In my resluts folder there are only 2 files 
"results/autopilot_longest6.json"
"results/transfuser_longest6.json"

bigbigbabyer-ai, 2025-09-22:
> > 删除 result.json 文件，代码会认为所有路由都已完成。您可以查看之前出现过几次的问题。
> 
> 我也遇到了类似的问题。 请问您能指定 result.json 的路径吗？我的结果文件夹中只有两个文件 ：results/autopilot_longest6.json 和 results/transfuser_longest6.json。

I deleted all of them, and after running the command, the transfuser_longest6.json file was generated. You can try deleting it and then running again.